seemoo-lab/openhaystack #3: Raspberry Pi Support

schoeberlt, 2021-03-04:
Hi there,
awesome work. I would like to play around and implement the BLE Advertisement on a Raspberry Pi. I'm happy to contribute to this project and open up a PR after implementing it successfully (and if it's okay with you)
Cheers
Tamas

schmittner, 2021-03-04:
RPi support is coming _very soon_!

DrSchottky, 2021-03-04:
@schmittner what about the desktop app? Is there any chance it could be ported to other operative systems or can't be untied from OS X?

minlite, 2021-03-04:
> @schmittner what about the desktop app? Is there any chance it could be ported to other operative systems or can't be untied from OS X?

Probably not. The plugin uses Apple Mail's entitlements to access an API. At the very least you would need a Macbook to extract the API key from.

schmittner, 2021-03-04:
> Probably not. The plugin uses Apple Mail's entitlements to access an API. At the very least you would need a Macbook to extract the API key from.

Correct, we query Apple's server with certain authentication HTTP headers, which we currently generating/retrieving via a private API. If we ever figure out how they are produced (the code is highly obfuscated), we might be able to provide an "owner" app on a non-Apple platform.

closed 2021-03-04 by schmittner

acalatrava, 2021-03-06:
Did you reverse engineer the Find app on the iPhone? Would be possible to add the items there?

Sn0wfreezeDev, 2021-03-06:
No, we do not add anything on Apple's servers or databases. Therefore, we should not invalidate any terms and conditions. 
Currently we do not offer an iPhone app as it would need to be jailbroken.

niyazaki, 2022-06-04:
Is it planned to have a docker app for the server part ? Of course we have to own a macOs device to get the private key once in a while. But could be super convenient to have it on a NAS.

commit a4ca840f (2021-03-04): Add generic HCI script (fixes #3)