seemoo-lab/openhaystack #5: Curious about the finding device

gbdavidx, 2021-03-05:
I use a similar method to track my apple watch aka myself for automations, whats the fastest response time when detecting a device such as an apple watch?

schmittner, 2021-03-05:
There is a delay between a finder device (iPhone) detecting a nearby tag and uploading a location report to Apple. We measured this in our paper (see figure below). The median is less than 26 min. In my personal setup with OpenHaystack at home, the reporting delay is usually below 5 min, though. Might have to do with the fact, that iPhones connected to Wi-Fi will upload reports to Apple more frequently as they don't have to be concerned with using up cellular data plan.

[image]

gbdavidx, 2021-03-05:
Thanks for the response. Seems to long for my use case but interesting
project. Would be interested to see if uwb can be applied here

closed 2021-03-10 by schmittner