seemoo-lab/openhaystack #6: Are microbits v2 compatible?

sagz, 2021-03-05:
There's a new micro bit in town, so would like to know if it's supported/compatible

Sn0wfreezeDev, 2021-03-05:
We have tested it just on the first version of micro:bits. 
Both are using a similar Nordic chip so the firmware theoretically should run on v2 as well. If you have one, you can try it out and send is feedback if it works.

schmittner, 2021-03-05:
The v2 micro:bit uses a newer chip from Nordic, so the current firmware will _not_ work.

It should be possible to port this to the nRF5 SDK to support more/all Nordic chips.

yonnath, 2021-03-18:
Hi, thanks. I was enthusiastic to test OpenHS but I have a micro:bit v2. Looking forward to testing a port for the nRF5 platform.

hakunti, 2021-03-24:
Wow, great job! 

If someone is interested to port firmware to RuuviTags and other devices running on nRF52, we will ship free RuuviTags and a the Nordic nRF52 development kit. Our code is available at https://github.com/ruuvi. Our software and hardware both are open-source :)

Br. Henri, Ruuvi

teutat3s, 2021-03-28:
I would like to help porting - contacted Ruuvi via email just now.

hakunti, 2021-05-08:
> I just published an alternative firmware that works on nRF52. I tested it
> myself on a E104-BT5032 module but @hakunti <https://github.com/hakunti>
> some free RuuviTags for testing won't hurt ;)
>
>
> https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative

Just confirm your address :)

Kind regards,

*Henri Hakunti*

*COO, Ruuvi Innovations Ltd (Oy)ruuvi.com <https://ruuvi.com>*

mhv75, 2021-08-11:
Hi @hakunti hoping to implement the development I am doing with nrf52 based smartwatches/fitness trackers. Would love to take ruuvi tags along in development. Very much aligns with the end goals I have in mind for the hardware. Does the offer still stand?

koenvervloesem, 2021-08-11:
Maybe interesting to know: I'm running my own Zephyr-based OpenHaystack firmware on both the micro:bit v2 and the RuuviTag: https://github.com/koenvervloesem/openhaystack-zephyr