seemoo-lab/openhaystack #9: Firmware for the esp32

greatjack1, 2021-03-05:
Does anyone have a version of this that can run on an esp32, either via a native image or via an arduino sketch?

greatjack1, 2021-03-08:
Looking at the code, I think I can build a basic script using the python hci script in this repo.

Will keep everyone updated.

schoeberlt, 2021-03-08:
My ESP32 Controllers are on their was, I‘ll try to do it using ESP-IDF. I‘ll check back soon

fhessel, 2021-03-08:
We now have an IDF-based firmware for the ESP32 which seems to work, it just needs a bit more testing. I'll create a PR once that's done.

ivale, 2021-03-10:
Hy I managed to build the esp firmware but I am not sure how to manage "public-key-in-base64". 
I tried to generate a key from the app and deliver to the ESP but FindMyIphone doesn't find the item. Can you help?

Sn0wfreezeDev, 2021-03-10:
We do integrate any devices to Find my on iPhone or macOS. 
You can only find your devices using our Mac App OpenHaystack. The next release will allow you to flash your ESP32 directly from your device.

schmittner, 2021-03-10:
Fixed in 898563ca0b2042f15d1a44f5efdb273af7d58ba0. Thanks @fhessel and @Sn0wfreezeDev 👍

closed 2021-03-10 by schmittner