seemoo-lab/openhaystack #11: Incompatible Plug-ins Disabled

froodproton, 2021-03-06:
After activating the plugin in mail, it restarts and shows:

Mail has disabled the following plug-ins:
OpenHaystackMail.mailbundle
Contact the makers of these plug-ins for versions that are compatible with Mail 14.0.

sem1845, 2021-03-06:
https://github.com/seemoo-lab/openhaystack/issues/4

froodproton, 2021-03-06:
I cannot confirm.
Maybe because:
sudo spctl --master-disable
objc[10597]: Class SPExecutionPolicy is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class AppWrapper is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class AppWrapperPolicyResult is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class AppWrapperPolicy is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class SPLog is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class MIS is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class SPExecutionHistoryItem is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class SPExecutionPolicyItem is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class SPDeveloperPolicy is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.
objc[10597]: Class GKScanResult is implemented in both /System/Library/PrivateFrameworks/SystemPolicy.framework/Versions/A/SystemPolicy and /usr/sbin/spctl. One of the two will be used. Which one is undefined.

??

Sn0wfreezeDev, 2021-03-07:
As suggested by @sem1845 
Disable Gatekeeper, then activate the Mail plug-in and enable gatekeeper again afterwards

froodproton, 2021-03-07:
I don't get the point - the gatekeeper was disabled before this issue was posted. 
Doesn't the status show that it is disabled?
sudo spctl --status                        
assessments disabled

After activating the plugin in mail, it restarts and shows:

Mail has disabled the following plug-ins:
OpenHaystackMail.mailbundle
Contact the makers of these plug-ins for versions that are compatible with Mail 14.0.

Sn0wfreezeDev, 2021-03-08:
Which macOS version are you using? It could be that a new Beta would require an update to the Plug-In

schmittner, 2021-03-08:
Also I think that running `spctl` shouldn't throw these warnings...

schmittner, 2021-03-15:
@froodproton were you able to get it running?

domse, 2021-03-29:
I think the Mailplugin isn't compatible to current Mail 14 version as per  openhaystack/OpenHaystack/OpenHaystackMail/Info.plist Therefore the plugin doesn't show up without another compatible Addon. I was also only able to install when adding mailbutler.

froodproton, 2021-04-08:
I am running regular Mac OS 11.2.3
The Mac is company managed with Jamf, but I don't see any restrictions regarding gatekeeper.

grigorye, 2021-05-01:
I face this issue as well on M1 running BigSur 11.3.

PrinzEric, 2021-07-22:
Still present

K4CZP3R, 2021-07-24:
Bump, still present on 11.5 (m1)