seemoo-lab/openhaystack #13: Flash common Aliexpress nRF51822

francescofact, 2021-03-07:
Hi, i was wondering what i need to flash pre built nRF51822 chips that can be found on alixpress.
I have a few of these laying around home for a little project and it would be awsome to use them as trackers.

Item: https://it.aliexpress.com/item/32885909449.html

Does anyone know what tool i need and how to mount them as volumes so i can copy the file like the microbit tutorial?

Sn0wfreezeDev, 2021-03-07:
Hi, 

we ordered some of those, but we have not received them yet. 
We are planning to do a small port if it is possible, but I think it would not be as easy as copying the firmware.

francescofact, 2021-03-07:
Fantastic!

Do you know what tool is needed to interface a pc/mac with them?

schmittner, 2021-03-08:
You should be able to re-flash them with a SEGGER J-Link or similar by connecting it to the SWD ports (they are labeled in the link that you provided).

francescofact, 2021-03-08:
So this should work?
https://it.aliexpress.com/item/32669702891.html

schmittner, 2021-03-08:
> So this should work?
> https://it.aliexpress.com/item/32669702891.html

Yes, it looks like it _should_ do the trick. No guarantee, though ;-)

schmittner, 2021-03-11:
This guy (#20) already did it using an ST-Link/V2. Good luck!

closed 2021-03-11 by schmittner

KaDw, 2021-03-12:
There is also  dirt cheap NRF24L01+ but it is able to send only 32 bytes of data. For advertising messages, this means maximum of 21 bytes of payload can be sent (32 - 6 (MAC) - 2 (header) - 3 (CRC)). I don't have any Apple devices, is there any other way I can test this solution?

Sn0wfreezeDev, 2021-03-17:
It won't work without a Mac for now. 
32 bytes of Bluetooth data is enough, because the address is part of the key