seemoo-lab/openhaystack #15: TILE support

mrx23dot, 2021-03-08:
Please add support for TILE network.
It runs on android.

https://www.thetileapp.com/en-gb/how-it-works
PCB: https://www.youtube.com/watch?v=k-BTStjaxgw

schmittner, 2021-03-08:
We won't add support for other tracking networks.

Leaving this open intentionally for others to see.

xenon149, 2021-04-13:
Hey @schmittner I think that @mrx23dot was probably wondering whether it is possible to implement the firmware on Tile hardware, rather than supporting their tracking network.

schmittner, 2021-04-13:
I don't own any Tile trackers and I don't know their hardware platform. Therefore, I wouldn't know how to flash a custom firmware onto them. Also, why flash OpenHaystack on a 30$ device if you can buy Nordic chips for 5$, which are already supported?

So, even if @mrx23dot was referring to the tracker hardware, the answer stays the same: **wontfix** 😃

greatjack1, 2021-04-14:
@schmittner Can you post some links to the 5$ nordic chip which is supported?

schmittner, 2021-04-14:
@greatjack1 Anything based on nRF51288 should work. Just browse related issues: #20 or #35.

devnoname120, 2023-08-06:
Looks like Tile Mate from 2021 is based on [nRF52810](https://www.edn.com/teardown-tile-mate-bluetooth-tracker-relies-on-software/).

mrx23dot, 2025-11-27:
nRF added support for Tile
https://docs.nordicsemi.com/bundle/nrf5_SDK_v17.1.1/page/group_tile_lib.html