seemoo-lab/openhaystack #20: Unable to get key into "OFFLINEFINDINGPUBLICKEYHERE!" for nRF51822 firmware

Z-pi, 2021-03-09:
Hey thank you for this great project! i have an issue with making the firmware for nRF51822, i don't know how to enter the public key in to the firmware. `/* don't make `const` so we can replace key in compiled binary image */
static char public_key[28] = "OFFLINEFINDINGPUBLICKEYHERE!";` I tried to paste the advertisement key (Base64) from the open haystack app there, this doesn't work the make commands gives this error `error: initializer-string for array of 'char' is too long [-Werror]` What's the right way to get the advertisement key from the app into the main.c file for the nRF51822. Thank you for answering the question!

Sn0wfreezeDev, 2021-03-10:
Hi, 

The key needs to be in raw data. So you would need to paste a byte array there instead of using the base64 encoded version. 

```
static char public_key[28] = {0x01,0x02,0x03,0x4,...};
```
To convert your key from Base64 to hex I can recommend [cyber chef](https://gchq.github.io/CyberChef/)
We just use this string to replace the content on demand in the already compiled firmware version inside the OpenHaystack app.

closed 2021-03-10 by Sn0wfreezeDev

Z-pi, 2021-03-10:
Thank you for your help! I got it working on a low cost nrF51822 beacon from aliexpress. I flashed the chip using a ST-link v2 and  now with the right public key it's working!
[image]

acalatrava, 2021-04-20:
> Thank you for your help! I got it working on a low cost nrF51822 beacon from aliexpress. I flashed the chip using a ST-link v2 and now with the right public key it's working!
> [image]

May I know how long your battery lasted? I did the same thing on the same beacon but changing the advertising interval to 5 seconds in order to get a device that should last months or even years according to the specs. However after 11 days the battery is now dead... 

I used an (apparently) good battery (Varta) and I measured battery voltage almost every day:
3.27V 8-04-21
2.87V 9-04-21
2.92V 10-04-21
2.89V 11-04-21
2.75V 15-04-21
2.69V 16-04-21
2.50V 18-04-21

So either the device is not going to deep sleep as it should or the battery was not good enough or there is something on this beacon that is also consuming the battery. I would like to know what was/is your experience with it.

I'm also creating a new firmware but using the SoftDevice from Nordic instead of Blessed. So far it seems to work but I still didn't put it on a battery.

Thank you!

Sn0wfreezeDev, 2021-04-20:
Hi @acalatrava, 

yes the blessed firmware is just an implementation without any device sleep state. We are currently working on a Nordic based firmware for the nRF52 family and it uses a lot less energy which should keep the devices alive for up to one year with a 210mAh cell battery. 

I currently mostly use the micro:bit and a 6.000mAh battery pack, which lasts for 2-3 weeks. This is not really portable, but my Ali beacons have not arrived yet.

acalatrava, 2021-04-20:
That’s good to know. I really thought that the current firmware was using some sleep mode though. 

I’m also working on a firmware using Nordic SDK11 with SoftDevice that should be compatible with both nRF51 and nRF52. When it’s working (probably tomorrow :)) I could share it. 

I also just received a very tiny nRF52 module where I want to put this. But I need a good firmware with heavy sleep feature to make it possible since the battery I’ll use is a CR1220 which is VERY small.

Sn0wfreezeDev, 2021-04-21:
Which module did you order? Could you share a link here?

acalatrava, 2021-04-21:
Of course! It’s a E104-BT5032 from EBYTE. I ordered at AliExpress from this link and received in just 8 days 😳 from China to Spain 

https://a.aliexpress.com/_vEHSWf

I still need to try to solder it since it’s very tiny let’s see if I’m able to!

mowtschan, 2021-04-21:
> Thank you for your help! I got it working on a low cost nrF51822 beacon from aliexpress. I flashed the chip using a ST-link v2 and now with the right public key it's working!
> [image]

Could you share the link for beacon from AliExpress here?

acalatrava, 2021-04-21:
> > Thank you for your help! I got it working on a low cost nrF51822 beacon from aliexpress. I flashed the chip using a ST-link v2 and now with the right public key it's working!
> 
> Could you share the link for beacon from AliExpress here?

I got that beacon from this link
https://a.aliexpress.com/_v0mnNZ

And used this to flash it
https://a.aliexpress.com/_u8bId5

Z-pi, 2021-04-26:
> > Thank you for your help! I got it working on a low cost nrF51822 beacon from aliexpress. I flashed the chip using a ST-link v2 and now with the right public key it's working!
> > [image]
> 
> Could you share the link for beacon from AliExpress here?

I got my beacons from here: https://www.aliexpress.com/item/Long-range-waterproof-programable-tracker-BLE4-0-NRF51822-beacon-for-iphone-and-for-Android/32917064741.html?spm=a2g0s.9042311.0.0.27424c4d89bPFv

> May I know how long your battery lasted? I did the same thing on the same beacon but changing the advertising interval to 5 seconds in order to get a device that should last months or even years [...]

My battery didn't last long, I changed the broadcast interval to around 1-2 sec in the firmware and I stil couldn't get the battery life past a few weeks on a CR2320 cell.