seemoo-lab/openhaystack #21: Power while Trakking

PrinzEric, 2021-03-10:


Sn0wfreezeDev, 2021-03-10:
The micro:bit currently sends out the signals as often as possible. This will drain more battery than needed to discover it by iPhones. I cannot really tell you how long a battery pack keeps the micro:bit alive as we did not exactly measure this. 
The one that was active for the longest time was using about 3000mAh in 5 days.

PrinzEric, 2021-03-10:
Thanks vor this Information Sn0wfreeze

Sn0wfreezeDev, 2021-03-21:
Hi @PrinzEric
I just wanted to inform you about my current test. I attached the BBC micro:bit to a 6000mAh battery pack and it continues to run for 2 weeks now.

PrinzEric, 2021-03-21:
[image]

i use it for my Segway Ninebot Max Scooter, Connected internal @5V DC Port from the Controller. Works very nice.

gekooo, 2021-03-21:
I Love this project! I have an nrf and it consumes 1mah. wanna try to reduce the advertisement frequency but i dont know how to do that. its been 5 days and the 3v battery pack is 2.7v now

PrinzEric, 2021-03-21:
I love it too ...
The BBC also have a CR2032 Battery Modul but I think it's empty to quick. For me with the Scooter is perfect.
Also car or use a 18650Battery Case u can use it very long time autark. 
But I think the official AirTags are releasing next days ;)

RayPS, 2021-07-03:
How can I set a lower frequency to save power?

PrinzEric, 2021-07-03:
Good Question, only Sn0wfreezeDev can make this. Maybe 1 or 5 or 10 Minutes.

Sn0wfreezeDev, 2021-07-06:
I'd recommend the firmware by @acalatrava. His firmware is linked here: https://github.com/seemoo-lab/openhaystack/issues/57#issuecomment-872807642 

It should last about 3 years. We are currently working on an enhanced firmware, but its not finished yet.