seemoo-lab/openhaystack #22: "Manage Plug-ins..." doesn't show up in mail app.

glowingkitty, 2021-03-10:
Installed the mail plugin, restarted the mail app - even restarted the mac and followed this guide (https://help.mailbutler.io/help/manage-plug-ins-button-is-gone-from-apple-mail) - but still no "Manage Plug-ins..." button is showing up...
Did Apple change anything mail plugin related in the most recent version of macOS? Or on Apple M1 processors?
Using the MacBook Pro 13 (M1), macOS 11.2.2.

Z-pi, 2021-03-10:
I got it working by installing an other mail plug-in om my M1 Mac. I installed the plug-in mailbutler which made the option "Manage Plug-ins" show up, the open haystack plugins can then be enabled.

sthorley, 2021-03-11:
I too had this issue. M1 Mac mini, OS 11.2.3. I tried working around this by opening mail in rosetta which had no effect. I also ran the commands mentioned in the MailButler article with no effect.
After installing the MailButler plugin as per Z-pi's comment above I was able to see the manage plugins button in both Apple Silicon and Rosetta mode so I don't believe this issue is M1 related.

glowingkitty, 2021-03-11:
> “OpenHaystackMail.mailbundle” can’t be opened because Apple cannot check it for malicious software.

Ok, button shows up, but cannot activate the plugin...

glowingkitty, 2021-03-11:
And this... [image]

glowingkitty, 2021-03-11:
#11

closed 2021-03-11 by glowingkitty

lkoehl, 2021-03-14:
> I got it working by installing an other mail plug-in om my M1 Mac. I installed the plug-in mailbutler which made the option "Manage Plug-ins" show up, the open haystack plugins can then be enabled.

Make sure to check #14 so you don't need to install another plugin.

sthorley, 2021-03-14:
> Make sure to check #14 so you don't need to install another plugin.

if you read the first post and my post you would realise that that command is part of the mailbutler troubleshooting steps I already performed.