seemoo-lab/openhaystack #33: M5StickC support

liucoj, 2021-03-17:
Hi,
I own an M5stickC that as you may know is an ESP32 with a little LCD,a battery and many sensors installed that should make a tag very interesting.
I successfully deployed manually on my MacOS your Firmware for ESP32 on my M5stickC, I added also an espressif-components during build for some specific support (enabling Led blinking when tag is in use) and I'll carry on deploying some other features for my M5 (printing Infos on LCD as battery status or something useful).
Do you think to support M5 hardware on a future release?

Sn0wfreezeDev, 2021-03-17:
I did not know about the M5, but it looks like an interesting model for creating actually practical tags. Have you tested how long the internal battery lasts? 

If you are able to get it running on the M5, which should be possible if its based on an ESP32, then I don't see any reason why we should not support it. You can add a pull request with your firmware code in the /Firmware folder and we can see if we can integrate it nicely

liucoj, 2021-03-17:
I tested the battery yesterday, I took M5 (100% battery) with me during my whole working day and after 8 hours it was still powered on. 
It sounds great but, of course, we have to consider that it was running only your firmware with its onboard red led blinking every 20 sec (i included M5stickC.h library on your .c file).

Anyway I'll make a pull request soon, I need only to sum up my ideas about :-D

Just for reference, you'll find here some about M5StickC and its features:
https://m5stack.hackster.io/products/m5stickc-esp32-pico-mini-iot-development-board
https://m5stack-store.myshopify.com/collections/m5-hat

bettse, 2021-05-11:
It is rather unclean, but here is a repo using platformio for an M5 Stick-C: https://gitlab.com/bettse/m5-needle

xynetta, 2021-06-13:
@liucoj Cool. I just got started tinkering this morning, got a bunch M5Stack ATOM Matrix kicking about. Sadly I'm no developer (SysAdmin) but I'll like mess around with stuff. I have not got my first one working yet, it flips from green to amber all the time and no location yet. When I noticed you added M5stickC.h you got me thinking. I should add support to my silly 5x5 matrix for statuses etc. Should be fun. Idéa is to plant the little bugger on my electric moped for tracking and perhaps have it to cut the power if needed. :) The Matrix comes with a push button so it would be nice to use it for a push sequence instead of the ignition key.

liucoj, 2021-06-17:
@xynetta I run OHS successfully on my M5StickC on both esp/IDF and Arduino IDE.
Not much efforts to port main. c from IDF to Arduino. 
What is easier on Arduino without doubts is implementing M5 functionalities like LCD, Sensors, Battery status or else.
Check out my repo https://github.com/liucoj/OHS-Arduino, it's a simple sketch but it's fully working.
In the next few days, I'll try to include some code to enable HAT's (speaker HAT),  battery icon and buttons.
This should run on your M5Stack because the code is basically the same.

[image]