seemoo-lab/openhaystack #35: Aliexpress nRF5182 beacon support

sthorley, 2021-03-26:
From reading #20 it seems I could use this on the Aliexpress nRF5182 beacon chips by simply taking the existing firmware.bin and replacing the string with the advertisement key.

Could this process be added to the app which could then just dump the patched firmware somewhere to manually flash?
I'm still awaiting my ST-link v2 to arrive for the actual flashing process so haven't tested things yet but the form factor of the nRF5182 chips is perfect for this application.

Sn0wfreezeDev, 2021-03-30:
Hi, 
The current firmware may run on the nrf5182 beacons, but it would use too much energy for the small batteries attached. 
I'd recommend that you have a look at the firmware code and reduce the sending interval at least. Currently the chips are sending as fast as possible, which will use up energy quite quickly. 

Nevertheless, I like the idea of copying the final firmware with the key to a location from which you can manually flash any chip.

acalatrava, 2021-04-02:
Is there any (good) reason why the current firmware is sending out the data as such fast rate?

schmittner, 2021-04-03:
No good reason. It's just an artifact from early testing.

Will update this soon.

froodproton, 2021-04-08:
I am running regular Mac OS 11.2.3 
The Mac is company managed with Jamf, but I don't see any restrictions regarding gatekeeper.

sthorley, 2021-04-28:
Got mine updated and working today. I'll do a quick write-up for anyone who was a bit intimidated by the flashing process (I was at the start)... This was tested on Mac mini M1 running Big Sur 11.3

1. Create New accessory in OpenHaystack app
2. Click Deploy and select "Export Microbit firmware"
3. Save the firmware file.
4. Solder 4 wires onto the 4 pads as shown in the image below.

[image]

5. Connect these wires to the corresponding pins on your STLink v2 and connect to Mac.
6. Install "Open-ocd" using Brew: `brew install open-ocd` If you don't have telnet installed install it too. `brew install telnet`
7. From terminal run: `openocd -f /opt/homebrew/opt/open-ocd/share/openocd/scripts/interface/stlink-v2.cfg -f /opt/homebrew/opt/open-ocd/share/openocd/scripts/target/nrf51.cfg`
8. Open-OCD should now be running
9. Open a new terminal window and use telnet to connect to the open-end session `telnet localhost 4444`
10. Send halt command: `halt`
11. Optional: Backup existing NRF Firmware: `dump_image /Your/File/Path/flash.bin 0 0x40000`
12. Erase existing flash: `nrf51 mass_erase`
13. Verify new firmware file: `program /Your/Firmware/File/Path/openhaystack_firmware.bin verify`
14. Program the firmware: `program /Your/Firmware/File/Path/openhaystack_firmware.bin`
15. Send resume command: `resume`

Congrats your tag should appear in OpenHaystacks momentarily.

mowtschan, 2021-05-03:
@sthorley Thank you! 

I just got it working on NRF51822 as well - it's amazing!!! Thanks!!!

just in case if someone will not find open-ocd configs, I found them here:
```
/usr/local/share/openocd/scripts/interface/stlink-v2.cfg
/usr/local/share/openocd/scripts/target/nrf51.cfg
```

yoution, 2021-07-04:
hi, @sthorley  thanks you
I meet this problem, can you help me?
[image]

mowtschan, 2021-07-04:
@yoution ,  Can you try to put the full path to the `openhaystack_firmware.bin`?

yoution, 2021-07-04:
@mowtschan thanks ,it's done, it's the file path error

yoution, 2021-07-04:
I use the framework directly, it seems the fresh rate is too low,I took it with my iphone in my car, and the points in the map 
 are too sparse  
[image]

Sn0wfreezeDev, 2021-07-07:
That's not really a firmware issue. It's more part of the design of Apple's Find My network. 
The device needs to be found by nearby iPhones and nearby iPhones won't report more than 4 locations of the same device in 30min to 1 hour. 
So if your device is moving you need A) a nearby iPhone and B) frequently changing nearby iPhones, because your own iPhone would not send a location update to Apple every minute or so. 

Tracking actually moving device with their exact paths does only work in very populated dense environments, like a shopping area of a town. Also it helps if you have a high density of iPhone users nearby. So 15-20% is probably a good enough rate.

yoution, 2021-07-07:
@Sn0wfreezeDev thanks, learn a lot
but I found another question,
[image]
I used a new battery that midday， but next day I took it outside， I found the location and time did not update, it was still the last night status
why did that happen? out of battery?

buzali, 2021-11-24:
Has anyone changed the beacon interval to make the battery use more efficient on these devices?

Sn0wfreezeDev, 2021-11-24:
Yes, please check @acalatrava repository for that: https://github.com/acalatrava/openhaystack-firmware

MoutardeForte, 2022-06-10:
Hello I have program with programming (I have a jlink)
[image]

Thanks for all

jrusi, 2022-08-15:
What’s the battery life with standard firmware on this specific device?

vbetker, 2022-08-19:
With standard firmware it's a couple of weeks, with acalatravas version its way longer - I have been running a couple of nRF51822s for half a year now without depleting the battery on any of them.

Itheras, 2022-08-19:
I can help you out just let me know.

Itheras, 2022-08-19:
> @jrusi I got about 5 days out of the default firmware. @vbetker could you perhaps help with flashing the firmware? [acalatrava/openhaystack-firmware#15](https://github.com/acalatrava/openhaystack-firmware/issues/15)

I can help you out just let me know.

Itheras, 2022-08-19:
what operating system and what issues are you having?

bruvv, 2022-08-19:
@Itheras https://github.com/acalatrava/openhaystack-firmware/issues/15 see that ticket please

Itheras, 2022-08-19:
> @Itheras [acalatrava/openhaystack-firmware#15](https://github.com/acalatrava/openhaystack-firmware/issues/15) see that ticket please

aaah yeah sorry let me see.

Paulnguyenbeam, 2022-09-18:
just a note you can put all the commands in the openocd command 

if you put
-c init -c "reset halt" -c "nrf51 mass_erase 0" -c reset -c "program fullpathtoyourbinfile verify"  -c reset -c exit
at the end of your command then it does it all without having to open a second termnial issuing the commands via telnet.  

Useful if you are doing a bunch at once

HarryWindsor, 2022-10-26:
> Hello I have program with programming (I have a jlink) [image]
> 
> Thanks for all

I am also only using a jlink I had to run `openocd -f /opt/homebrew/opt/open-ocd/share/openocd/scripts/interface/jlink.cfg -c "transport select swd" -f /opt/homebrew/opt/open-ocd/share/openocd/scripts/target/nrf51_mod.cfg`. Where I added `set WORKAREASIZE 0` to `nrf51_mod.cfg`.

But I cannot see my device online. How did you flash it with jlink?

backupandforkacc, 2023-01-25:
Is there a programming adapter like there is for blitzwolf and tasmota?
https://www.thingiverse.com/thing:3476167

devnoname120, 2023-08-06:
@dzianisv What battery life do you get with this beacon?
https://aliexpress.com/item/1005003671695188.html?sku_id=12000028906157906

dzianisv, 2023-08-08:
> @dzianisv What battery life do you get with this beacon? https://aliexpress.com/item/1005003671695188.html?sku_id=12000028906157906

with CR2025 it works about 14 days, looks like the firmware has to be optimized, for instance a deep sleep has to be used? But I am not very familiar with NRF chips internals.

rkreutz, 2023-11-04:
Hey everyone, I've tried flashing the firmware onto the nRF51822 ali express beacon as suggested from above, but I had no success, running the following code returns an error
```
openocd -f /usr/local/share/openocd/scripts/interface/stlink.cfg -f /usr/local/share/openocd/scripts/target/nrf51.cfg -c "init; reset halt; nrf51 mass_erase; program firmware.bin verify; program firmware.bin; resume;"
```
I made sure to run this command on the same directory as `firmware.bin` is located, which is a manually compiled firmware with the appropriate pub key, I get the following error message:
```
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.300599
Info : [nrf51.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf51.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x00000c70 msp: 0x20004000
Info : nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Info : Mass erase completed.
Info : A reset or power cycle is required if the flash was protected before.
[nrf51.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Info : Padding image section 0 at 0x00003d5f with 1 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x00003d60 .. 0x00003fff
Error: jtag status contains invalid mode value - communication failure
Error: error waiting for target flash write algorithm
Error: Failed to enable read-only operation
Error: Failed to write to nrf5 flash
Error: error writing to flash at address 0x00000000 at offset 0x00000000
embedded:startup.tcl:1516: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 1581
at file "embedded:startup.tcl", line 1516
``` 
Any help would be much appreciated, many thanks.

solarkraft, 2024-08-28:
Hey, sorry if this is too off topic. 
I’m thinking about putting this thing into some hardware and am wondering about the measurements, but can’t find any officially. 
It seems to be designed to fit on top of a CR2477, which would be around 24.5mm in diameter. 
Can anyone confirm this?
Thank you!

lovelyelfpop, 2024-11-27:
Can I patch the firmware with Linux or Windows, I have no MacOS

kleuter, 2024-11-27:
Yes, openocd is available for all major platforms

ghd182, 2025-01-13:
I recently purchased an nRF51822 beacon from AliExpress and, after a day of trial and error, successfully flashed the firmware. It’s now working flawlessly! I performed the firmware flashing using a Raspberry Pi 4 with Dupont wires soldered to the beacon.
Since I'd like to experiment with the firmware outside my house without having to solder it again and again I’m curious if it’s possible to enable wireless firmware updates via OTA (Over-the-Air) or DFU (Device Firmware Update) for this device. If so, could you provide any guidance or resources on how to implement this feature?

Thank you all for your incredible work and the wealth of knowledge you’ve shared so far... it’s been a huge help!

acalatrava, 2025-01-14:
> I recently purchased an nRF51822 beacon from AliExpress and, after a day of trial and error, successfully flashed the firmware. It’s now working flawlessly! I performed the firmware flashing using a Raspberry Pi 4 with Dupont wires soldered to the beacon. Since I'd like to experiment with the firmware outside my house without having to solder it again and again I’m curious if it’s possible to enable wireless firmware updates via OTA or DFU for this device. If so, could you provide any guidance or resources on how to implement this feature?
> 
> Thank you all for your incredible work and the wealth of knowledge you’ve shared so far... it’s been a huge help!

Some time ago I developed a firmware with DFU support, but it's for nRF52. Maybe you can use it as a guide to create it for the nRF51 family: https://github.com/acalatrava/openhaystack_moko

lovelyelfpop, 2025-05-20:
@jessienab I have tried 1, 2 and many other 51822/52810/52832 ble modules or devices, pix's firmwares works good on all of these boards.  Option 2 use CR2477 battery, the size of this board is bigger than option 1(cr2032)

HieuDAY1, 2025-12-16:
Open On-Chip Debugger
> halt
> nrf51 mass_erase
nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Mass erase completed.
A reset or power cycle is required if the flash was protected before.
> program D:/openhaystack-fw/openhaystack-firmware/apps/openhaystack-alternative/compiled/nrf51_firmware_patched.bin
[nrf51.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x00001c84 .. 0x00001fff
** Programming Finished **
> resume
[nrf51.cpu] clearing lockup after double fault
[nrf51.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0xc1000003 pc: 0xfffffffe msp: 0xffffffd8
[nrf51.cpu] Polling failed, trying to reexamine
[nrf51.cpu] Cortex-M0 r0p0 processor detected
[nrf51.cpu] target has 4 breakpoints, 2 watchpoints
[nrf51.cpu] Examination succeed

Can someone tell me what this error is?