seemoo-lab/openhaystack #36: Build an iOS app for tracking at any time

shadowsss, 2021-03-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Sn0wfreezeDev, 2021-03-27:
Hi,

We cannot build an iOS App by now, because on iOS we would not get access to the necessary tokens to download the locations of your devices. It would only work on  jailbroken iPhones.

shadowsss, 2021-03-28:
You can use altstore to distribute the ipa package to install on the iPhone device.

Sn0wfreezeDev, 2021-03-28:
That would allow us to install the app, but to download locations on the iPhone you would need to have access to a token that is stored in the keychain. The item is protected by an entitlement, which can only be used if the iPhone is jailbroken.

shadowsss, 2021-03-28:
You can log in to appleid in the app to get the token

Sn0wfreezeDev, 2021-03-28:
This is not possible.
We would have considered it, if it would be technically possible.

vpceg, 2021-04-11:
There are plenty of jailbroken devices in the world. The community has been raving about the app.

Sn0wfreezeDev, 2021-04-12:
Hi @vpceg, 
yes it should not be very complicated to port this app to an iOS app. Most of its code is written ins SwiftUI and it should be easy to reuse it on iOS. 
Only the code that is currently living in the Mail Plug-In needs to be ported, which shouldn't be too complicated. As the code is open source, we are also open to Pull requests with an iOS version. 😊

Sn0wfreezeDev, 2021-04-15:
Okay I had some time today to go for an idea of a simple self hosted server combined with an iOS app. 

You can start the server on your local Mac, expose it to the internet with ngrok and use the iOS App to show where your devices are.  
The code is available here: 
https://github.com/Sn0wfreezeDev/openhaystack/tree/simple_server

Please see this as a first draft. I don't know if I have more time in the near future to shape it.

ivale, 2021-04-21:
Can you explain how to use it pls?

Sn0wfreezeDev, 2021-04-21:
Sure. As none of this is bundled or finished, you will need to clone the branch *simple_server* on the repository I linked above.   
Then you can open the OpenHaystack Xcode project. It contains two new targets: An iOS app and an OpenHaystackServer. 
You can start by running the server, it will automatically host attach on port 8080. Then to make this server accessible from the outside (e.g. the iPhone app) you will need to forward the connection (either by opening a port in your router firewall) or by using a service like ngrok. I'll quickly explain ngrok, because it is simpler to use. 
After you installed ngrok and you are running the server, you can launch it with: `ngrok http 8080`. This will assign a randomly looking https url to you. 

Now you can start running the iOS app on your iPhone. The iOS app has a text field which can be used to enter your own Server url. Using Universal Clipboard you can copy the URL on your Mac and paste it on your iPhone. Then you can import your accessories from the Mac app (export them as a plist first).  After importing the iPhone will start fetching the location reports over your server running on your Mac. This allows you to track your devices anywhere. 
There are some obvious downsides: 
- Your Mac needs to be awake to answer requests 
- The app is not available in the App Store and not as an IPA to use it with AltStore (yet)

Loic760, 2021-09-08:
> Sure. As none of this is bundled or finished, you will need to clone the branch _simple_server_ on the repository I linked above.
> Then you can open the OpenHaystack Xcode project. It contains two [...]

I'm getting this error on my server **{"error":true,"reason":"Not Found"}** (http://127.0.0.1:8080)
I'm running original version of mail plugin + open haystack and only compiling the simple_server app.

Does the "open haystack simple_server" + "simple server plugin" are mandatory for the simple server to run ? or could I use the "normal" ones ?

Furtivo360, 2021-12-01:
Hi! This project is amazing: really well done!
I've been able to flash 2 beacons but I am not able to understand how to see those from my iPhone FindMy app.
I can see the 2 devices from the Mac OpenHaystack application and I wanna be able to monitor the items from everywhere.
Can someone explain it to me please?
Thanks in advance 

Diego

Sn0wfreezeDev, 2021-12-02:
@Furtivo360 the iOS is very basic and does not really work permanently. 
We are working on an iOS and Android version, but we need to make sure that the access to the reports is stable

Furtivo360, 2021-12-02:
@Sn0wfreezeDev thanks for your reply :)
Keep up the good job! 👍

davesenior9, 2023-03-06:
Hey Guys,

Im trying to use Simple Server following the above guidance based on; https://github.com/Sn0wfreezeDev/openhaystack/tree/simple_server, however, i cant even see in the source the mention on the simple_server anymore. Ive also had a look at similar questions and cant see anything helpful.

Additionally, when I try to install Xcode on Big Sur, i'm faced with 'Xcode can't be installed on "Macintosh HD" because MacOS version 12.5 or later is required.'

Is there a more detailed guide for setting up simple_server?

My end state is interaction with lat, long, time, device id/name in a python program. 

Thanks