seemoo-lab/openhaystack #37: Private keys of regular devices

acalatrava, 2021-04-05:
I’m trying to use a public key from one of the devices currently registered on Find My App. This way I could create a tag that should appear as one of my Macs. According to the docs the private key should be on the iCloud Keychain but I’m unable to find it. 
Does anybody knows where is it and how can I export it?
Thanks!

Sn0wfreezeDev, 2021-04-06:
This is technically possible, but quite hard to implement. 
The keys are not directly located in the macOS keychain. The keychain contains a key called `BeaconStore`, which is used to encrypt the private keys synchronized between multiple devices. Those private keys are called `Master Beacon` and they are located at `~/Library/com.apple.icloud.searchpartyd/MasterBeacons/`. The *.record files are property lists, which contain another property list. To decrypt them you should use the `Beacon Store` key and AES GCM. 

After you've got the actual private key you cannot just use its public key to public this one and have your device appear in FindMy. For this your would need to adapt the firmware that is currently available to send rotating keys as mentioned in our paper (see Section 6.1) 

We are currently working on a firmware with rotating keys, but we do not have an ETA, yet.

acalatrava, 2021-04-06:
Thank you for your input! I understand that you can't use the public key directly, but I don't think you need to implement the whole rotating keys thing, I guess that if you generate the first key and the device send that first key continuously that should work.

Also I tried to check those keys on my keychain but they are not present on any of my Macs. I'm still running MacOS Mojave so maybe that's the cause since offline finding appeared on MacOS Catalina, but my iPhone is running iOS14 so I thought at least their they should be there... but maybe is not synced since I'm not on Catalina?

BTW, I maybe able to help with the new firmware if you need to. I'm trying to port the firmware to Go (using TinyGO) although it seems that the crypto packages are not available for nrf51822 so maybe that's not possible.

I'm not sure if this is the right communication channel or you have slack or something where we can discuss this technical things in a faster way.

Sn0wfreezeDev, 2021-04-06:
This is definitely the best communication channel for the moment. 

I think generating the first key does not work, because from our reverse engineering we know that the devices are only checking for current keys (for the last 7 days). You could probably generate one current key and deploy it to one of your devices. This could work for about 7 days. 

The reason why you cannot find the keys is definitely macOS Mojave. You would need Catalina for Offline Finding and even Big Sur for Open Haystack. 

I'll keep this issue open for further discussions

llama, 2021-06-03:
> To decrypt them you should use the Beacon Store key and AES GCM.

@Sn0wfreezeDev how do you get the BeaconStore key? When I run `security find-generic-password -l 'BeaconStore' -g` no value is printed. Thanks.

Sn0wfreezeDev, 2021-06-03:
I find it in `KeychainAccess` easily

llama, 2021-06-03:
Odd, I have multiple airtags showing up in Find My, but the password field is blank for the BeaconStore item in KeychainAccess. Perhaps they relocated the key?
[image]

Sn0wfreezeDev, 2021-06-04:
The content might be in any raw data that cannot be represented by some text. Probably because the randomly generate those passwords. You can probably access the data with Swift and Apple's security framework.

llama, 2021-06-04:
I did also try accessing it from Swift and saw no data stored under that keychain item. 

When arbitrary data is stored as the value, the KeyChain app does not allow checking the "Show Password" box, and `security` shows some representation of it still with the `-g` flag.  For example, you can see OpenHaystack's keychain plist data with `security find-generic-password -l 'FindMyAccessories' -g`.

Sn0wfreezeDev, 2021-06-07:
Sorry, I don't have any source code available for this. The last time I decrypted my private keys was with macOS Catalina. Apple has changed a lot there after we submitted a security vulnerability.

llama, 2021-06-11:
Alright, sorry for all the posts here but if anybody else is trying to decrypt these we can compare results. When I call `SecItemCopyMatching` I get the following item back:

```
Optional({
    acct = BeaconStoreKey;
    cdat = "2021-06-05 00:45:09 +0000";
    class = genp;
    gena = {length = 32, bytes = 0xf6f63d5f <redacted> c8274547 };
    labl = BeaconStore;
    mdat = "2021-06-05 00:45:09 +0000";
    svce = BeaconStore;
    "v_Data" = {length = 0, bytes = 0x};
})
```

Note the empty data, but the promising looking "gena" or "kSecAttrGeneric". I tried using that value to decrypt the .record files: `openssl enc -aes-256-gcm -d -in 1B0AC147<credacted>6C187AC41D6E.record -out test.out -K f6f63d5f<redacted>c8274547 -iv 0` but this fails with `bad decrypt`.

I also tried using the "gena" value from the very similar looking LocalBeaconStore, but with the same failure.

llama, 2021-06-11:
Perhaps my issue is that I don't have the correct IV. The .record property lists have three values. The first two are short and the last one is long. I've been trying to decrypt the long value using the value under the "gena" attribute of the BeaconStore keychain item as the key, and one of the short values from the .record plist as the IV. Neither of them worked, nor did an IV of 0.

llama, 2021-06-11:
Yay I was finally able to decrypt these .record files. For anybody else trying to do this:

1. Open the .record files as plists. 
2. First value is the nonce. Second value is the tag. Third value is ciphertext.
3. The key is the  "gena" or `kSecAttrGeneric` of the BeaconStore keychain item. You can get it's hex encoding very simply with `security -v find-generic-password -l 'BeaconStore' -g`.
4. Decrypt using `AES.GCM.open(sealedBox, using: key)` in `CryptoKit`. The result is a plist.

I had been stumped by the fact that there was no proper data associated with the keychain item in step (3), along with trying to use `openssl` to decrypt the file rather than cryptokit. Thanks again @Sn0wfreezeDev.

robinkruyt, 2021-06-28:
Did any of you get any further with this?
I’ve successfully decrypted the plist but have ended up with a private, public, sharedsecret and sharedlocationsecret key. 
None of which seem to be usable to generate valid rolling keys.

Sn0wfreezeDev, 2021-06-29:
Hey @robinkruyt, 

Congrats! That's exactly what you need to generate the keys. 
Please refer to the paper in section 6.1 to see how the rolling keys are generated. 
d = private key, p = public key, SK = shared secret.

robinkruyt, 2021-07-04:
@Sn0wfreezeDev Thanks for confirming which keys are what.
It seems that I am unable to generate valid public keys, but it's hard to verify whether that is because I generate invalid keys, or because the advertisement keys I've gotten are incorrect ( I'm trying to capture them using a bluetooth sniffer )

Any change you could direct me to some code that generates keys?

Secondly, I'm not that knowledgeable on crypto in general, but when section 6.1 talks about `KDF(SKi−1, “update”, 32)`, does that mean to use the "update"-string as the shared info?

ps. I could create a new issue if this is considered issue hijacking

luke-jr, 2022-11-21:
>Open the .record files as plists.

`plutil -convert xml1 XXXXX.record` seems to work

>The key is the "gena" or kSecAttrGeneric of the BeaconStore keychain item. You can get its hex encoding very simply with security -v find-generic-password -l 'BeaconStore' -g.

That command yields nothing. But without the `-g` it shows a "gena" blob...

>Decrypt using AES.GCM.open(sealedBox, using: key) in CryptoKit. The result is a plist.

Is there a way to actually do this without installing development tools?

GijsGoudzwaard, 2022-12-06:
In my `~/Library/com.apple.icloud.searchpartyd` I do not have a folder called `MasterBeacons`, though I did see that folder on another macbook. Is there a certain setting I need enable in order for apple to generate the `.record` files? I'm logged into my Apple ID on my Macbook. I've tried to relog into my Apple ID on macbook. I do see that when I log out of my Apple ID  and log back into it that the directories `BeaconNamingRecord`, `CKTokens`, `LeashRecords`, `OwnedBeacons`, and `SecureLocationInfo` are removed and created again in my `~/Library/com.apple.icloud.searchpartyd` folder.

Does anyone know how to get the `MasterBeacons` folder?

PaulBerry-Briggs, 2024-01-04:
Did anyone get any further with this?  

I'm trying to do the same thing as @leokeba, track airtags that I own using standard ble scanners.  

I understand from the discussion above that to do so I need to extract the private key for the airtag from my icloud keychain (by using the technique posted by @llama above) and then generate a public key from that private key (or maybe generate a set of possible public keys?) and then compare those to the public key being broadcast by the airtag, which is split across the ble address and the ble manufacturer data.... but it *might* be enough to just compare the approriate bits from the generated public keys to the bits in the ble address.

Does all that sound about right?

YeapGuy, 2024-01-28:
>     4. Decrypt using `AES.GCM.open(sealedBox, using: key)` in `CryptoKit`. The result is a plist.
How exactly would I go about this? I tried a few things, but I can't even get my code to compile...

denysvitali, 2024-06-28:
I have created this:
https://github.com/denysvitali/searchparty-keys

Given the `.record` files (and a `BeaconStore` password) - it provides a way to get all the necessary keys - which can then be used to find your lost items.

thisiscam, 2024-12-04:
On macOS 15 running `security -v find-generic-password -l 'BeaconStore' -g` gives 
```
find-generic-password "-l" "BeaconStore" "-g"
security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.
```

crschmidt, 2024-12-11:
> On macOS 15 running `security -v find-generic-password -l 'BeaconStore' -g` gives
> 
> ```
> find-generic-password "-l" "BeaconStore" "-g"
> security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.
> ```

From experimenting, it seems that Apple may have moved the BeaconStore into the iCloud/Local Items keychain, which (as far as I can tell) can not be accessed from the `security` tool or any other tool. You can probably confirm this by opening up the Keychain Access app and searching for BeaconStore to see if the BeaconStore key is located on the `login` keychain or the iCloud or "Local Items" keychain. If so, this seems like it makes it much more difficult to extract keys from OS X 15.x, which I would guess is *probably* intentional on Apple's part.

thisiscam, 2024-12-12:
> > On macOS 15 running `security -v find-generic-password -l 'BeaconStore' -g` gives
> > ```
> > find-generic-password "-l" "BeaconStore" "-g"
> > security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.
> > ```
> 
> From experimenting, it seems that Apple may have moved the BeaconStore into the iCloud/Local Items keychain as part of MacOS 15, which (as far as I can tell) can not be accessed from the `security` tool or any other tool. You can probably confirm this by opening up the Keychain Access app and searching for BeaconStore to see if the BeaconStore key is located on the `login` keychain or the iCloud or "Local Items" keychain. If so, this seems like it makes it much more difficult to extract keys from OS X 15.x, which I would guess is _probably_ intentional on Apple's part.

Yes, the BeaconStore is indeed now in the iCloud keychain. I also recently found that Apple might have changed the key rotation algorithm on official airtags: https://github.com/malmeloo/FindMy.py/issues/90 So maybe this is part of their attempt to strengthen security of official accessories. Unfortunately, that means I can't use official airtags anymore for my own purpose..

thisiscam, 2024-12-12:
Here's another plausible explanation of this: https://github.com/MartinPham/FindMySync/issues/25#issuecomment-2266719318 .

Basically, I'm stuck at fetching the password to decode the plists because of this.

mobilityvalley, 2025-02-05:
So there is no way to use this repo to track locations ?