seemoo-lab/openhaystack #39: Compatibility with Apple's Find My app.

StefanNienhuis, 2021-04-08:
Awesome project! 

I was just wondering, now that Apple has released the 'Items' tab in the Find My app, does this project work with it or could it be made compatible?

Bubba8291, 2021-04-09:
+1
I would like to see this too

schmittner, 2021-04-09:
TL;DR: _In principle_, this could work. _In practice_, it won't.

There are essentially two ways that we could implement this.
**Option 1**: We extract the advertisement keys from the genuine Find My app and display all devices in our OpenHaystack app (this is what #37 suggests).
**Option 2**: We inject the advertisement keys of our OpenHaystack devices into Apple's Find My app.

Both options would require us to access the secured storage (e.g., keychain) of the Find My processes. To do this on macOS, you need to **disable system integrity protection (SIP)**, which would leave your system vulnerable to all sorts of attacks. This is nothing that you should do on a production machine. And this is why we won't go down that road. Also, we would need to conduct more manual reverse-engineering to figure out where to inject/from where to extract the keys.

For completeness sake: Certified third-party Find My devices use a proprietary pairing process involving Apple-issued certificates to appear in Apple's Find My app.

StefanNienhuis, 2021-04-09:
Thanks for the detailed response.

schmittner, 2021-04-14:
I'll leave this open so others can easily find it

mrx23dot, 2021-04-21:
Interesting, the site below says:
- the app generates ONE private key per user
- and multiple public keys, which are assigned to devices

https://news.ycombinator.com/item?id=20129942

So chances are it assigns public keys in plain text. (no point encrypting)
And the tag simply broadcasts that public key.
The mac will need to connect to the tag for assignment. The type of the tag I guess is simply identified by the BLE MAC with a random id part. (I guess it will work offline, so it wouldn't check for copycat tags)
An ad said it does this automaticaly for connected earpods.

Would be interesting to record the BLE traffic for assigning an original airtag and see the broadcast at the end. Does anyone have one?

If Apple took the shortest path everywhere then we can easily mass produce airtags.

mrx23dot, 2021-04-21:
Also it would be easy to trigger a sound from any nearby tags since a 3rd party user can also do this (stalking mode). It would be easy to destroy the whole apple airtag brand :dagger:

Sn0wfreezeDev, 2021-04-22:
To me the text hacker news post basically looks like the person paraphrased our paper about the [Find My network](https://arxiv.org/abs/2103.02282v1). 

The main difference for OpenHaystack for now is that our firmware and apps do not change the advertised public keys and it seems like AirTags send a slightly different BLE advertisement. The rest is almost identical and that's why OpenHaystack actually worked before AirTags have been released. 

I plan on buying some AirTags so analyze their Bluetooth packets.

mrx23dot, 2021-05-04:
If you have some wireless accessory (like earpods) you can already use the finding feature for those.

schmittner, 2021-05-12:
@throwaways this would work in principle as I mentioned. But quoting myself:

> Also, we would need to conduct more manual reverse-engineering to figure out where to inject/from where to extract the keys.

llama, 2021-06-03:
> Both options would require us to access the secured storage (e.g., keychain) of the Find My processes. To do this on macOS, you need to disable system integrity protection (SIP)

I have SIP enabled. When I run `security find-generic-password -l 'BeaconStore'`, I'm able to see a hex value under "gena". Is that not the AES GCM decryption key? 

If not, is there really no way to extract it from the keychain without disabling SIP? Even if so, as long as this key does not change it should only be necessary to obtain once, then SIP could be reenabled.

Thanks for your awesome work!

parawanderer, 2025-04-13:
I ran into this thread while searching for something else. 

FYI to readers: 
The python library [malmeloo/FindMy.py](https://github.com/malmeloo/FindMy.py) supports official AirTags. And you can get the private keys without doing any insane levels of security disabling via scripts like the following so long as you use MacOS version `<= 14`:

- [`airtag_decryptor.py`](https://github.com/parawanderer/OpenTagViewer/tree/main/python#airtag_decryptorpy) -- the same script used by the MacOS Export Wizard. 
- [`airtag-decryptor.swift`](https://gist.github.com/airy10/5205dc851fbd0715fcd7a5cdde25e7c8) by [airy10](https://gist.github.com/airy10) on which the above script is based
- [`airtag-decryptor.swift`](https://gist.github.com/YeapGuy/f473de53c2a4e8978bc63217359ca1e4) by [Matus](https://gist.github.com/YeapGuy)

The security implications of this are that you will need to give the scripts access to your `BeaconStore` key. So, nothing crazy like turning off all security on MacOS.