seemoo-lab/openhaystack #43: Deploy firmware on ESP32 fails

mowtschan, 2021-04-14:
I would like to ask for help, If I run `./flash_esp32.sh -p /dev/cu.usbserial-0001 "my public key in base64"`,
I got:

```
esptool.py v3.0
Found 4 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........__
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XXXXXXXXX
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.1 seconds.
Staying in bootloader.
esptool.py v3.0
Found 4 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.usbserial-0001 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header```

As I understand "erasing" of the device works fine, but then "writing" fails. Am I missing something?

fhessel, 2021-04-14:
The script tries to keep the ESP32 in bootloader mode between erasing the NVM partition and writing the firmware, which seems not to work in your case.

As you already get console output, I assume you are able to run the script on its own. Could you try replacing lines [132-138](https://github.com/seemoo-lab/openhaystack/blob/acdae59b3985be18c054c2140d688582646d7fc7/Firmware/ESP32/flash_esp32.sh#L132-L138) with the following to see if uploading as such, without explicit bootloader management, works? (basically removing the erase an the `--before no_reset`)

```bash
esptool.py --baud $BAUDRATE \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
```

Also, it would be good to know what specific ESP32 hardware you are using. By default, esptool uses the CTS and DTR lines to control the bootloader, but they have to be wired correctly and the USB-to-Serial interface has to support that.

mowtschan, 2021-04-15:
@fhessel Thanks for the fast reply! ❤️

>  Also, it would be good to know what specific ESP32 hardware you are using.

ESP32 hardware: 
https://smile.amazon.de/gp/product/B074RGW2VQ

I tried what you suggested and it works! 
Did it solve the problem or I cannot use the device like that?

```
esptool.py v3.0
Found 4 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........___
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: xxxxxxxxxxxxxxx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 23488 bytes to 14467...
Wrote 23488 bytes (14467 compressed) at 0x00001000 in 1.3 seconds (effective 146.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 113...
Wrote 3072 bytes (113 compressed) at 0x00008000 in 0.0 seconds (effective 1673.4 kbit/s)...
Hash of data verified.
Compressed 732 bytes to 594...
Wrote 732 bytes (594 compressed) at 0x0000e000 in 0.1 seconds (effective 102.7 kbit/s)...
Hash of data verified.
Compressed 617776 bytes to 386733...
Wrote 617776 bytes (386733 compressed) at 0x00010000 in 34.1 seconds (effective 144.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

fhessel, 2021-04-15:
I have the same (AZ Delivery) boards lying around here, and I remember having trouble with the bootloader on that particular model in the past. I'll try to see if I can reproduce your issue using that board and will get back to you.

> I tried what you suggested and it works!
> Did it solve the problem or I cannot use the device like that?

If the firmware launches successfully, it will work (you could check that on the serial console, e.g. with `miniterm.py /dev/cu.usbserial-0001 115200`. You may need to press the reset button after flashing before you get any output). The reason for erasing the region is that the ESP32 might not boot if it contains an old, corrupted NVM partition. That can happen even if flashing the actual firmware succeeds. So it's just a precaution, because we don't know what was running on the modules before.

mowtschan, 2021-04-15:
Looks good right? 
` miniterm.py /dev/cu.usbserial-0001 115200`

```
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
...
␛[0;32mI (974) open_haystack: using device address: ...............␛[0m
␛[0;32mI (984) open_haystack: application initialized␛[0m
␛[0;32mI (1004) open_haystack: advertising has started.␛[0m
```

fhessel, 2021-04-15:
Yes, that's how it should look like when the application is running. Otherwise you would see some errors about nvm.

mowtschan, 2021-04-15:
Is it possible to add that device to the OpenHaystack App?

fhessel, 2021-04-15:
In principle, it should work like for every other ESP32 board. I tried it several times with the AZ Delivery board, but that works for me (but that bootloader problems I had earlier with this hardware were not deterministic, either).

```
Wrote 617776 bytes (386733 compressed) at 0x00010000 in 34.1 seconds (effective 144.7 kbit/s)...
```

Did you run the script with the `--slow` option (and did you have a reason for that)? Otherwise, that step should take only 5-6 seconds. If it falls back to the slower rate by itself, there might be other issues with the USB-to-Serial interface as well.

The only "workaround" that I could think of is swapping both steps (meaning to flash first, then erase), and to do a hard reset and sleep in between. That would then look like this:

```bash
esptool.py --baud $BAUDRATE \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
sleep 2
esptool.py erase_region 0x9000 0x5000
```

You could also try that one (again replacing the same lines mentioned above) – that should be a version that prevents nvm issues but does not rely on staying in bootloader between steps.

Sn0wfreezeDev, 2021-04-15:
> Is it possible to add that device to the OpenHaystack App?

You can create new devices in OpenHaystack and deploy them manually by right-clicking on the entry in the table and select _copy advertisement key_. Then you add this key to your ESP board by flashing (as done before). The device should appear in the OpenHaystack app

mowtschan, 2021-04-15:
> Did you run the script with the `--slow` option (and did you have a reason for that)?

Yes, I tried the script with `-s` flag as well, I thought maybe flashing didn't work because of the fast baud rate.

 The "workaround" work! Thanks a lot! ❤️
```
...
Wrote 617776 bytes (386733 compressed) at 0x00010000 in 5.7 seconds (effective 870.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v3.0
Found 4 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........_
Detecting chip type... ESP32
....
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.2 seconds.
Hard resetting via RTS pin...
```

mowtschan, 2021-04-15:
@Sn0wfreezeDev, @fhessel Thank you, guys! ❤️

[image]

closed 2021-04-15 by mowtschan

brightproject, 2023-07-05:
> WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
> Crystal is 40MHz

What is it?