seemoo-lab/openhaystack #44: Add "cleanup" to the deployment script

mowtschan, 2021-04-14:
**Is your feature request related to a problem? Please describe.**
If you try to deploy firmware with `./Firmware/ESP32/flash_esp32.sh`
and it fails, you need to remove `tmp.key` file manually before you can run the script again.

**Describe the solution you'd like**
I would like to have a kind of "cleanup" in `./Firmware/ESP32/flash_esp32.sh`:
```
cleanup() {
    echo "cleanup ..."
    rm -v "$KEYFILE"
}
...
set -e
trap cleanup INT TERM EXIT
...
```

Sn0wfreezeDev, 2021-04-15:
At the moment the deployment script is mainly built to run from our app, which always removes the entire directory when finished. 
You can make a pull request with this function for the manual installation and I'm happy to accept it.

mowtschan, 2021-04-15:
Should be fixed with PR #46