seemoo-lab/openhaystack #49: AirTag privacy vs OpenHaystack

acalatrava, 2021-04-22:
Checking the specs from the upcoming AirTag you can read this:

> AirTag is designed to discourage unwanted tracking. If someone else’s AirTag finds its way into your stuff, your iPhone will notice it’s traveling with you and send you an alert. After a while, if you still haven’t found it, the AirTag will start playing a sound to let you know it’s there.

My guess is that the AirTag will broadcast another advertisement besides the OF advertisement but I’m not so sure about that. Maybe this is a new thing on iOS 14.5 that will alert also when a custom OpenHaystack device is around? (I don’t think so since the OF protocol will change the advertisement every 15 minutes) 

Any thoughts on this?

Sn0wfreezeDev, 2021-04-22:
I have the original documentation for third party Find My devices on my computer, which was detailing on the BLE messages. As far as I remember the messages just have one byte that differs from the current message in OpenHaystack. With the current 14.5 Beta I did not get any tracking notifications, but it might be that they will appear with the release version. 

Otherwise, we can change the byte to see if this enables the stalking protection. But this could be annoying, because your own iPhone would react to this as well, because it does not know your OpenHaystack tags

archoversight, 2021-05-13:
@Sn0wfreezeDev would you be able to share that documentation?

In the paper on page 5, the following is mentioned:

> Initially, each owner device generates a private–public key pair (d0,p0) on the NIST P-224 curve and a 32-byte symmetric key SK0 that together form the master beacon key. Those keys are never sent out via BLE and are used to derive the rolling advertisement keys included in the BLE advertisements.

and d0 is used to generate derived keys di, pi, where pi is the public part that is advertised.

However in the README.md it states that the public key is shared with the AirTag, not the private: https://github.com/seemoo-lab/openhaystack#pairing-1

Is d0 the private key or the public key?

All other uses of `d` seems to be the private key, hence the confusion.

Sn0wfreezeDev, 2021-05-17:
The specification that you cited from our paper is the one that is used in lost iPhones and MacBooks. 
This is not identical to the OpenHaystack tags, because the current version just creates a private-public key pair and no symmetric key SK0. The keys do not rotate and therefore, we share the public key with the OpenHaystack tag. 

The private key stays in your keychain.

archoversight, 2021-05-19:
> The specification that you cited from our paper is the one that is used in lost iPhones and MacBooks.

Yes, I am somewhat hijacking the current topic of conversation, would be happy to create a new ticket instead if you'd like.

There is still confusion in the pdf regarding d0 vs p0 and which one is shared with the AirTag, in the paper it specifies d0 is shared, even though in the paper it states that it is the private part of the private-public key part.

Hence my question.

Sn0wfreezeDev, 2021-05-19:
Could you cite the section in which the paper says that d0 is shared and not the public key? In the current version  it says `The BLE advertisements sent out by a lost device contain an EC public key pi`.  I don't really get where the confusion is coming from so it would be great if you could go into more detail here. 

For AirTags, I cannot say if they get the private key as this has not been reverse-engineered by us. They shouldn't because they don't need it. You can generate rolling public keys just by using SK0 and the first public key p0.

Itheras, 2021-12-26:
Slight update I have seen airtag tracking you messages since iOS 15 with open haystack tags . Maybe not as consistent but definitely some anti stalking functionality is working. I have a tag in my vehicle and if I drive for longs periods I will get alerted sometimes. For the tags at home no alerts. Its seems to only be triggered if a open hay tag follows me around for extended periods.

Sn0wfreezeDev, 2022-01-03:
Yes @Itheras I saw something similar with my Tag in the car as well. So iOS 15 has added the missing stalking detection here

zer0detail, 2022-08-18:
Hi @Sn0wfreezeDev, I think I am also confused by the same problem that archoversight mentioned regarding references to d0.

In your last comment you mentioned:

> You can generate rolling public keys just by using SK0 and the first public key p0.
this makes sense intuitively as the Air Tags also need to generate the rolling keys and they could be given access to sk0 and p0 while leaving d0 (the initial private key) off of the tracking device and safely stored in the owner device keychain.
It's also backed up by this [video](https://youtu.be/3byNNUReyvE?t=2072) which discusses the broadcast keys key derivation coming from "P".

The point of confusion comes from section 6.1 of the report under the **Master Beacon and Advertisement Keys** section.
Specifically in the 4 step equation to generate the new keys, step 3 tells us that the new private key di comes from using Ui and Vi with d0 (not p0)
SKi = KDF(SKi−1, “update”, 32) 
(ui, vi) = KDF(SKi, “diversify”, 72) 
**di = (d0 ∗ ui) + vi** 
pi = di ∗ G 

Thats followed up by the description:

>  Eqs. (3) and (4) create the advertisement key pair via EC point multiplication using the anti-tracking keys and the master beacon key d0.

Which once again says to use d0 to create the advertisement key.

I believe that is why @archoversight  was asking you if you were using possibly mistakenly using d0 to refer to the public key (in the report), or if you actually mean that the advertisement keys need the private key d0 to help generate them.