seemoo-lab/openhaystack #50: ESP32 AirTag-like board hardware requirements

ameeuw, 2021-04-25:
Hi all, since I haven't seen any type of esp32 based boards by the usual suspects (ttgo, lilygo, lolin) that would fit my personal requirements for an AirTag-like board my question is: What board are you guys using beyond development-boards?

Technically it could just come in the form of a battery holder that routes out the programming header from a micro32 board.

My ideal board would:
- run on CR2032 (maybe larger) cell
  - No WiFi RF startup shouldn't be a problem
  - Not sure how long the cell can provide up to 2.5V under load
- be based on
  -  led-less micro esp32 module [LilyGo Micro 32](https://de.aliexpress.com/item/32869180373.html) comes to mind
  - OR esp32 directly on board (much more effort to design specific board)
- ADC for voltage readout
- have header or test-pads (VCC, GND, IO0, TX, RX) to connect to programming base

Since this is not much, I am sure I forgot a lot... What do you guys think? Any dev board out there I am missing (round design? battery management included?)

Thanks for any ideas 😊

Sn0wfreezeDev, 2021-04-27:
Hi @ameeuw, 

It's great that you opened the issue here. On the ESP32 side I am personally only using development boards with attached battery packs or a larger lithium battery. I have not found a perfect ESP32 and those which I use are using too much energy. 

Some people here had success with nRF51822 chips that have a CR2032 cell holder attached. You can get them at Aliexpress here: https://de.aliexpress.com/item/32885909449.html?spm=a2g0o.productlist.0.0.1b1a182d87cSss
With the current firmware the batteries won't last very long though.

@schmittner  and I are currently testing on the next nRF generation, but this needs a new firmware. At the moment, we were not able to get it running reliable. 
https://de.aliexpress.com/item/4000389437486.html?spm=a2g0s.9042311.0.0.4bad4c4d1vdn2Z 

There are some other discussions about the nRF chips here: 
https://github.com/seemoo-lab/openhaystack/issues/21

Maybe @fhessel knows more about suitable ESP32 boards.

fhessel, 2021-04-27:
> Maybe @fhessel knows more about suitable ESP32 boards.

I don't have any boards matching that form factor at hand, only some experience in wiring an ESP32 WROOM module to a 18650 LiPo cell, but an 18650 is nothing I would consider small.

The main issue with a CR3032 cell is that its 3V are already close to the minimum input voltage of ESP32 modules with a 3.3V flash in them (cf. [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf), page 42), so one would probably want to use one that can go down to 2.3V for VCC or – as you mentioned – place the bare ESP32 directly to the board. All other stuff from typical development boards would have to go as well (voltage regulators, USB-to-Serial, status LEDs, …). But what I'm mostly concerned about would be the peak current provided by a coin cell. The ESP32 has some short peaks drawing a lot of current during radio startup (I've seen peaks exceeding even 500mA at 3.3V), and a supply that does not deliver will trigger the brownout detection, creating a boot loop. I'm not sure if that also applies for initializing BLE only, but that would make creating generic full-featured (meaning: WiFi-capable) ESP32 boards in coin cell size hard, which might be the reason why you don't find them with the usual suspects.

Creating such a tag would be interesting nonetheless.

RoganDawes, 2021-04-29:
FemtoBeacon looks like the closest thing at the moment, but it would be best with a buck-boost converter instead of the LDO, if it will run on a 2032 battery.

https://www.crowdsupply.com/femto/femtobeacon