seemoo-lab/openhaystack #53: Add device name

mowtschan, 2021-04-30:
**Is your feature request related to a problem? Please describe.**
I would like to see the name of the BLE device when discovering with some BLE discovering tools (e.g. BlueSee)

**Describe the solution you'd like**
Add device name to adv data?
https://github.com/seemoo-lab/openhaystack/blob/ffc5170ea4b4ceb1ad84e4f89324d6e666ffc7c3/Firmware/ESP32/main/openhaystack_main.c#L27

I tried to put the name with the `esp_ble_gap_set_device_name` function before 
https://github.com/seemoo-lab/openhaystack/blob/ffc5170ea4b4ceb1ad84e4f89324d6e666ffc7c3/Firmware/ESP32/main/openhaystack_main.c#L157

the function returns no error, but the device name is still not appear in the discovery tools :-/

schmittner, 2021-05-02:
Sending the name does not work. The reason is that the name is transmitted as part of the BLE advertisement payload – and there is simply no space left. The find my advertisement already takes all of the 31 bytes in the payload.

closed 2021-05-02 by mowtschan

ch0rus, 2022-01-20:
You can send the device name in the scan response