seemoo-lab/openhaystack #55: error when flashing esp32

jaekim24, 2021-04-30:
ESP32 % ./flash_esp32.sh -p /dev/cu.usbserial-0001 "yaAusUDjMLRkXaRqZ0TLTvWeM9cbW3fdCtxnQD/xySI="
esptool.py v3.0
Found 2 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........_
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:36:9c:ac
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.1 seconds.
Staying in bootloader.
esptool.py v3.0
Found 2 serial ports
Serial port /dev/cu.usbserial-0001
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.usbserial-0001 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.Bluetooth-Incoming-Port failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
cleanup ...

This is the error I get and idk what to do any advice would be appreciated

jaekim24, 2021-05-01:
I don't know if this is a reason why its not working or not but I noticed "performing build step for  'boot loader' " does not finish and it only does 2 out of the 4 tasks if supposed to do [](url)
[image]

mowtschan, 2021-05-01:
@jaekim24 check this https://github.com/seemoo-lab/openhaystack/issues/43, it worked for me.

jaekim24, 2021-05-01:
@mowtschan I know he said lines 132-138 but do you just replace what's under #Clear NVM on line 137 with what that guy said to replace to fix it 

[image]

mowtschan, 2021-05-01:
@jaekim24 this works for me:
```
esptool.py --baud $BAUDRATE --port "$PORT" \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
sleep 2
esptool.py --port "$PORT" erase_region 0x9000 0x5000
```

jaekim24, 2021-05-01:
@mowtschan it worked but when I try to deploy it it says it couldn't 
[image]

mowtschan, 2021-05-01:
@jaekim24 you don't need to deploy it with the App, because your esp32 device already has the firmware deployed (You did it in the previous step with the deploy script). In the app just right-click -> "Mark as deployed"

jaekim24, 2021-05-01:
ohh I did not know that thank you

jaekim24, 2021-05-01:
but when I run "miniterm.py /dev/cu.usbserial-0001 115200" I get an error saying 
no module named serial " so I install serial with pip and I already had serial 
[image]

jaekim24, 2021-05-01:
it works thank you @mowtschan

closed 2021-05-01 by jaekim24

mowtschan, 2021-05-01:
you can try it with `screen`:
```
screen /dev/cu.usbserial-0001 115200
```