seemoo-lab/openhaystack #56: implementing deep/light sleep

strik3, 2021-05-06:
I've been racking my brains for the past week trying to think how it would be possible to implement deep/light sleep into the esp32 firmware. this would drastically reduce power consumption but I'm a complete novice when using the esp ide. is this something you may be able to implement.

mowtschan, 2021-05-06:
@strik3 it should be not very difficult to implement, maybe this example can help (at least I would try it first):
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/examples/system/light_sleep/main/light_sleep_example_main.c

strik3, 2021-05-15:
managed to get it working. its not the prettiest or most efficient way of doing it. but it works and saves a ton of power.

`#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>

#include "nvs_flash.h"
#include "esp_partition.h"
#include "esp_bt.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"
#include "esp_bt_defs.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_sleep.h"
#include "esp_event.h"
#include "esp_timer.h"

static const char* LOG_TAG = "open_haystack";

/** Callback function for BT events */
static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param);

/** Random device address */
static esp_bd_addr_t rnd_addr = { 0xFF, 0xBB, 0xCC, 0xDD, 0xEE, 0xFF };

/** Advertisement payload */
static uint8_t adv_data[31] = {
	0x1e, /* Length (30) */
	0xff, /* Manufacturer Specific Data (type 0xff) */
	0x4c, 0x00, /* Company ID (Apple) */
	0x12, 0x19, /* Offline Finding type and length */
	0x00, /* State */
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0x00, /* First two bits */
	0x00, /* Hint (0x00) */ 
};

/* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_ble.html#_CPPv420esp_ble_adv_params_t */
static esp_ble_adv_params_t ble_adv_params = {
    // Advertising min interval:
    // Minimum advertising interval for undirected and low duty cycle
    // directed advertising. Range: 0x0020 to 0x4000 Default: N = 0x0800
    // (1.28 second) Time = N * 0.625 msec Time Range: 20 ms to 10.24 sec
    .adv_int_min        = 0x0021, // 10s
    // Advertising max interval:
    // Maximum advertising interval for undirected and low duty cycle
    // directed advertising. Range: 0x0020 to 0x4000 Default: N = 0x0800
    // (1.28 second) Time = N * 0.625 msec Time Range: 20 ms to 10.24 sec
    .adv_int_max        = 0x0021, // 10s
    // Advertisement type
    .adv_type           = ADV_TYPE_NONCONN_IND,
    // Use the random address
    .own_addr_type      = BLE_ADDR_TYPE_RANDOM,
    // All channels
    .channel_map        = ADV_CHNL_ALL,
    // Allow both scan and connection requests from anyone. 
    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};

static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    esp_err_t err;

[... 95 lines cut ...]
    ESP_LOGI(LOG_TAG, "application initialized");
    
    while (true) {
        // Wake up in 1.9 seconds
         
        esp_sleep_enable_timer_wakeup(1900000);

        ESP_LOGE(LOG_TAG, "Entering light sleep");

        // Enter sleep mode
         
        esp_light_sleep_start();
        
        // Execution continues here after wakeup
         
        ESP_LOGE(LOG_TAG, "Returned from light sleep");
        
        // Lets transmit a couple of times
        vTaskDelay(10);
    }
    
}
`

https://github.com/strik3/openhaystack/blob/main/Firmware/ESP32/main/openhaystack_main.c

strik3, 2021-05-15:
Managed to get it working. its not the prettiest or most efficent  but it saves a lot of power and gets the job done

https://github.com/strik3/openhaystack/blob/main/Firmware/ESP32/main/openhaystack_main.c

Sn0wfreezeDev, 2021-05-17:
Great that you added sleep to the esp32 firmware. 
I'll definitely give this a test run this week.

0raculo, 2022-03-20:
Any success with deep sleep?

thebigboss84, 2022-06-12:
Did you try to measure how long your device last with battery ?