seemoo-lab/openhaystack #57: Alternative firmware

acalatrava, 2021-05-07:
I just created an alternative firmware that is compatible with both nRF51 and nRF52 platforms. It's using the Softdevice S130/S132 2.0.0 from Nordic SDK11. 

I already tried it on this modules:
 - E104-BT5032A board from EBYTE which can be purchased here https://www.aliexpress.com/item/4000538644215.html
 - "AliExpress beacon" which can be purchased here https://www.aliexpress.com/item/32826502025.html

More info here:
https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative

mowtschan, 2021-05-07:
@acalatrava could you share the "howto" flash E104-BT5032A board? 
did you use a special board, adapter or just STLinkV2?

acalatrava, 2021-05-08:
Please note that you need to flash the SoftDevice too. You can generate a full bin file by issuing `make build`

Can you share your consume after that? I don’t have the tools to measure it.

acalatrava, 2021-05-08:
@mowtschan I used a JLink-compatible adapter, however you should be able to use the STLinkV2 too. I think it should work if you just issue `make && make e104install`, this will use `nrfjprog` to flash the firmware which I think it should work with STLinkV2 too.

junleus, 2021-05-08:
@acalatrava with softdevice nrf51_firmware.bin Size around 3.1MB, how to flash into nrf51 chipset? 

for openhaystack firmware standby 860uA, and sent advertisements will consume 960uA.

acalatrava, 2021-05-08:
You’re right. I think the problem is that those are not actually `bin` files but `hex` ones. I’m not at the computer atm so please try to rename it and flash it that way.

junleus, 2021-05-08:
the **bin** comes with the public key, but **hex** one doesn't have a public key. 

[image]

[image]

acalatrava, 2021-05-08:
Both bin and hex files are actually the same firmware on a different format. You can’t see the key on the bin file with a simple editor since the firmware is encoded in hexadecimal format (that’s why the file is bigger) instead of binary. 

So you can replace the key on the source file, compile, merge it with the SoftDevice and flash the resulting hex file (the compiled file appears as bin but it’s actually a hex file). 

I’ll try to fix this naming issue when I’m at home.

junleus, 2021-05-08:
@acalatrava  can you confirm that compiled nrf51_firmware.bin size is 3.1MB?

I try to flash with ST-LINK v2 but seem to run out of space. 

[image]

acalatrava, 2021-05-08:
No, it’s not a bin file but a hex one. You should try to rename it to hex
and then either flash it with nrfjprog (not sure if you can use it with
stlink) or convert it to bin (I think there is a hex2bin.pay tool somewhere)

junleus, 2021-05-08:
finally, I manual merge softdevice hex file got 324kb size hex,  flashed with STLINK v2, and keep consume 5uA, so no advertisements yet. I flashed back into openhaystack firmware and got advertisements again. I think need way hex2bin then.

appreciate.

mergehex -m s130_nrf51_2.0.0_softdevice.hex openhaystack-alternative_s130.hex -o softdevice.hex

jaekim24, 2021-05-09:
how long did your nRF52 chip  last with the coin cell battery @acalatrava

junleus, 2021-05-09:
I use ### objcopy cmd in Linux can convert hex to a bin, as the same way compiled an example s130-beacon apps it is running, but ### openhaystack-alternative doesn't run at all.

acalatrava, 2021-05-09:
@junleus I can confirm that something is broken with the nRF51 since it isn’t working on mine either. I’ll try to check what I broke, but since I have it working on another nRF51 for 7 days now it shouldn’t be hard to find. 

@jaekim24 I still didn’t put a coin cell on my module but since it’s working for more than 7 days on my nRF51 and the battery is still at 2.96V I expect a very long duration.

jaekim24, 2021-05-09:
what kind of battery did you use the on nRF51 @acalatrava

acalatrava, 2021-05-10:
I just published the fixed firmware http://github.com/acalatrava/openhaystack-firmware/ 
This one should work fine on a nRF51 module (tested!) 

@junleus please test it and tell me how much it consume. Also, how are you measuring it?

@jaekim24 I'm using this module https://www.aliexpress.com/item/32826502025.html with a CR2032 cell coin battery. Battery still at 2.96V :)

mowtschan, 2021-05-10:
@junleus unfortunately on my iPhone I'm unable to find my nrf51 but with the Macbook it works, I can even see exactly which device is mine:
1. copy adv. key:

[image]

2. run command and see from byte 7 which public key has your device(just 3-4bytes would be enough):

[image]

3. On mac I'm using BlueSee tool (https://apps.apple.com/de/app/bluesee-ble-debugger/id1336679524?mt=12):

[image]

acalatrava, 2021-05-10:
I’m pretty sure it’s working. The thing is that the advertisement is set every 5 seconds. Sometimes on the iPhone won’t appear after 10 or even 20 seconds. Probably you are seeing 5uA because most of the time is sleeping :)

try changing the advertisement interval or try this app which seems to work better to see the device on iPhone 
 https://apps.apple.com/es/app/ebeacon-ble-scanner/id730279939

mowtschan, 2021-05-11:
@junleus I bought it by AliExpress here: https://www.aliexpress.com/item/33061215469.html?spm=a2g0s.9042311.0.0.27424c4dgqLbvo
P.S.: just to clarify: I’m using original firmware from openhaystack.

acalatrava, 2021-05-11:
@junleus if the s130-beacon is working then my firmware should work too. Did you try to change the advertising interval? Try to change it to 100, that way you should see it on the light blue app or similar
https://github.com/acalatrava/openhaystack-firmware/blob/eef0a6b81e405b79027f0cf450c3e4b3322a76b3/apps/openhaystack-alternative/main.c#L10

acalatrava, 2021-05-11:
Which modules did you order?

you should try the firmware on the nRF51 during the day and check if the location is being updated on the app

junleus, 2021-05-11:
https://www.amazon.com/gp/product/B081GYNM2G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

jaekim24, 2021-05-12:
@junleus  I bought the same one and it arrived today!

jaekim24, 2021-05-12:
omg they're so tiny, I'm going to try to flash it today. Wish me luck

jaekim24, 2021-05-12:
im trying to flash the nRF52 but idk where to connect my VCC port from my J-Link debugger to the nRF52 module and also which ground am I supposed to connect to on the nRF52 module 
[image]

acalatrava, 2021-05-12:
You can use any GND because all of them are connected. VCC === VDD

jaekim24, 2021-05-12:
thank you !!@acalatrava

jaekim24, 2021-05-12:
@acalatrava how do I flash the nRF52 chip is it similar to how you flash a ESP32 with open-haystack like this [ ./flash_esp32.sh -p /dev/yourSerialPort "public-key-in-base64"] but a little different ?

mowtschan, 2021-05-13:
@acalatrava just got your custom firmware working on nrf51 device, power consumption is 3,3uA !!! with peak of 66uA, nice!!

[image]

acalatrava, 2021-05-13:
@jaekim24 if you have jlink you can just flash it issuing `make flash`

@mowtschan thats great! Very little consumption!!! If my math are correct a CR2032 220mah battery should last more than 1 year!

junleus, 2021-05-13:
> @acalatrava just got your custom firmware working on nrf51 device, power consumption is 3,3uA !!! with peak of 66uA, nice!!
> 
> [image]

@mowtschan  what kind of flash cable you are using?  I try few days and till got 5uA,  no advertisements send yet.

acalatrava, 2021-05-13:
@junleus look at this
https://github.com/acalatrava/openhaystack-firmware/issues/2#issuecomment-840596966

junleus, 2021-05-13:
awesome!

junleus, 2021-05-14:
so far so good. 
got  **5uA -> 25uA -25uA->200uA - 5uA.**

acalatrava, 2021-05-14:
Cool! Is this on nRF52 or nRF51? It is considerable more consumption than  @mowtschan  https://github.com/seemoo-lab/openhaystack/issues/57#issuecomment-840621399 which is weird...
Do you know for how long is consuming 25uA and 200uA?

junleus, 2021-05-14:
1. nRF51
2. Around every 6 seconds for 200uA.

mowtschan, 2021-05-14:
I have nRF51822-QFAA device:
3,3uA for <5 sec 
66uA for <1 sec (can't really see that value on multimeter because it appears for a very short time, just able to see it with slow-motion recording)

mowtschan, 2021-05-15:
> @mowtschan thats great! Very little consumption!!! If my math are correct a CR2032 220mah battery should last more than 1 year!

@acalatrava if my calculation not wrong then it would last for almost **3 years**?!?!?
Let's assume we have a battery with a capacity of 230 mAh and we will have ideal temperature of 20°C
and also let's 'ignore' battery self-discharging thing.
http://products.varta-microbattery.com/applications/MB_DATA/DOCUMENTS/DATA_SHEETS/DS6032.pdf

`(230/((0.0033*5+0.066*0.5)/(5+0.5)))/24 = 1064,814^` days!!!

Or am I missing something?

acalatrava, 2021-05-16:
Yeah! I think you’re correct! But I guess it would be a little bit less since it should be something like this 

`(230/((0.0033*4.5+0.066*0.5)/5))/24 = 1001,393`
Since the interval is set every 5 seconds it means that it will use 3,3uA for 4,5 sec and 66uA for 0,5 sec every 5 seconds... right?

Sn0wfreezeDev, 2021-05-17:
Thank you @acalatrava 
I tried the firmware today on an [nRF52832 Ali Express Beacon](https://de.aliexpress.com/item/4000389437486.html). 
It works great. Thank you for the effort to make it more energy efficient. 

Unfortunately, they increased the price to of the beacon 11$ recently.

mowtschan, 2021-05-17:
@Sn0wfreezeDev I bought 4x (but nrf51) here for 16€
https://de.aliexpress.com/item/33061215469.html?spm=a2g0s.9042311.0.0.7b414c4djbmG56
and now they are 22€ but still 50% less than nrf52

acalatrava, 2021-05-17:
@Sn0wfreezeDev great! Did you measure consumption?

Sn0wfreezeDev, 2021-05-18:
@acalatrava No I do not have the tools here to measure the consumption. But it is lasting for some days now, which would not be the case for the old firmware 

@mowtschan  Yes, but I wanted to go for the nrf52 to get the updated chip version. However, nrf51 should have all features needed for this application.

sebi5361, 2021-06-11:
@mowtschan Could you tell me what multimeter you are using? I am looking for one that can measure low currents (uA, nA) and was wondering if yours could do the job...

mowtschan, 2021-06-11:
@sebi5361, here you are https://smile.amazon.de/gp/product/B07DWG59F2/
```
Measuring range V/DC min. : 0.1 mV 
measuring range V/DC max. : 600 V 
measuring range V/AC min. : 0.1 V 
measuring range V/AC max. : 600V 
Measuring Range A/DC Min. : 0.1 µA 
measuring range A/DC max. : 10 A 
measuring range A/AC min. : 0.1 µA 
measuring range A/AC max. : 10 A
```

sebi5361, 2021-06-11:
@mowtschan Thank you! Good to know

sebi5361, 2021-06-13:
@acalatrava @mowtschan I would like to point you to [that video](https://www.youtube.com/watch?v=LUB8RWzzLWc) that focuses on measuring tiny currents. In a nutshell it says that a standard multimeter won't do the job properly due to its internal impedance resulting in a _burden voltage_.

sebi5361, 2021-07-01:
I just measured the consumption of OpenHaystack (ultra low power) alternative firmware on an **nRF51822** board equipped with a low frequency **crystal** ([YJ-14001 BLE module](https://www.aliexpress.com/item/1005002787181715.html)) using my freshly received [NORDIC Power Profiler Kit II](https://www.nordicsemi.com/Software-and-tools/Development-Tools/Power-Profiler-Kit-2).
This board's hardware is similar to the official [AliExpress Beacon](https://www.aliexpress.com/item/32826502025.html).

Below are the results:
## Full cycle:
[image]
## Broadcasting mode:
[image]
## Sleep mode:
[image]

According to those readings, the average power consumption is **7.85uA** which makes a **3V CR2032 230mAh** battery last more than **3 years**:
```
230/0.00785 = 29299 hours = 1220 days = 3.34 years.
```
Those results seem to be coherent with the multimeter measurements done by @mowtschan and the battery duration computed by @acalatrava .

sebi5361, 2021-07-01:
To compare current consumptions, I compiled the OpenHaystack (ultra low power) alternative firmware firmware for my [YJ-14001 BLE module](https://www.aliexpress.com/item/1005002787181715.html) (nRF51822 ) using the internal RC oscillator instead of the low frequency 32768 kHz external crystal oscillator.

Here is a comparison chart:

| Low Frequency Oscillator      | Average Current Consumption @ 3V |
| ------------------------------- | ---------------------------------------- |
| External 32768 kHz crystal    | 7.85 uA                                                |
| Internal RC circuit                  | 8.24 uA (14% more)                            |

This result corroborates [this](https://devzone.nordicsemi.com/f/nordic-q-a/14214/how-much-current-consumption-when-we-use-internal-32k-rc) external Crystal vs internal RC oscillator comparison discussion.

In a nutshell it is better to have a board equipped with a **low-frequency 32768 kHz external crystal oscillator** to save a little bit of power.

sebi5361, 2021-07-01:
I measured the consumption of OpenHaystack ultra low power alternative firmware on an **nRF52832** board equipped with **NO external low frequency crystal** ([Holyiot YJ-17095](https://www.aliexpress.com/item/32864571031.html)).

This board's hardware is similar to the official [eByte E104-BT5032A](https://www.aliexpress.com/item/4000538644215.html)'s hardware.

Thus I used `NRF_MODEL=nrf52 BOARD_E104BT5032A make build` to generate the firmware.
Note that *e104bt5032a_board.h* and *simple_board.h* are the same and correspond to boards with low frequency internal RC oscillators.

My board seems to broadcast correctly as it can be found in the OpenHaystack map.

Hower power consumption is over the charts compared to previous results:
**It is 37uA in average (4x more than for my nRF51822 board).**

Current consumption looks like this:
[image]
There is a spike every ~12.5ms.
Any idea why I observe such a high power consumption?

acalatrava, 2021-07-01:
I put the firmware on a E104-BT5032A one month ago and it seems that the battery is depleting quickly, so it seems this is also happening on this board.... not sure why though. Can you test with some other program?

sebi5361, 2021-07-01:
> Can you test with some other program?

With what other program? I'll do the test then.

sebi5361, 2021-07-02:
@acalatrava: If you send me firmwares for nrF52832 with NO crystal you are developing to solve this consumption issue, I can patch them with my advertisement key, test them and measure their current consumptions. Is this what you are meaning by "test with some other program" (firmware)?

sebi5361, 2021-07-08:
@acalatrava: BTW, did you enable the built-in DC/DC converter to save some energy (if the boards support it)? See [this explanation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fpower.html&cp=4_2_0_17_0&anchor=topic) and [this schematic](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fref_circuitry.html&cp=4_2_0_52_1&anchor=schematic_qfn48_dcdc).

sebi5361, 2021-07-14:
I compared the power consumption of my [Holyiot YJ-14001 BLE module (nRF51822)](https://www.aliexpress.com/item/1005002787181715.html) with different settings:

**AVERAGE POWER CONSUMPTION (in uA @ 3V):**

| Low-Frequency Oscillator / Internal Power Supply Regulator                 |  LDO (Default)     | DC/DC |
| ----------------------------------------------------------------------------- | ------------------ | -------- |
| Internal RC Circuit  (Default)                                                                     | 9.21                    |  9.64     |
|  External 32.768 kHz Crystal                                                                      | **7.60**              |  7.98     |

Note: to enable the DC/DC regulator I added line `sd_power_dcdc_mode_set(1);` before the while loop in `main.c`. This could be done as the Holyiot YJ-14001 BLE module is populated with the proper LC components required to enable this feature (see [Nordic InfoCenter](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fpower.html&cp=4_2_0_17_0&anchor=topic) for explanations). I don't know about the official [AliExpress beacon](https://www.aliexpress.com/item/32826502025.html).

Contrary to what I read, switching the regulator from LDO to DC/DC didn't improve power consumption.
The lowest power consumption was obtained selecting the **external 32.768 kHz crystal** together with the **LDO regulator**.

Below are screenshots of the measurements made using my [NORDIC Power Profiler Kit II](https://www.nordicsemi.com/Software-and-tools/Development-Tools/Power-Profiler-Kit-2) for the different settings.
Note the spikes due to clock calibrations here and there (see explanations from the [Nordic DevZone](https://devzone.nordicsemi.com/f/nordic-q-a/14214/how-much-current-consumption-when-we-use-internal-32k-rc)) when selecting the internal RC oscillator.

## N51-RC-LDO
[image]

## N51-XTAL-LDO
[image]

## N51-RC-DCDC
[image]

## N51-XTAL-DCDC
[image]

sebi5361, 2021-07-14:
And additional comment:
Not all [AliExpress beacons](https://www.aliexpress.com/item/32826502025.html) are populated with the low frequency external crystal.
[image]
Check it out before enabling the external low-frequency 32.768 kHz crystal.

junleus, 2021-07-14:
I checked this before, per the beacon do not need the 32k crystal.so all beacon device will not have this 32k crystal.

sebi5361, 2021-07-15:
Note the 4s interval spikes due to the clock calibration involved when using the internal low-frequency RC oscillator (see figure **N51-RC-LDO** blue rectangles) (orange rectangles correspond to 5s interval broadcasting spikes):
[image]
Those spikes are responsible for the extra average power consumption compared to enabling the external low-frequency 32.768kHz crystal (9.21uA vs 7.60uA).

Furtivo360, 2021-12-07:
Hi all! Good stuff in here, thanks for all your good work! 💯 👍
QQ: I am not really sure if I've done the right thing.

I've copied the main.c content from this file:
[image]

moved inside the openhaystack-main folder of the openhaystack application (see blow), pasted the "modded low power" main.c script inside the original one and saved.

[image]

Is my understanding correct that whenever I will now generate the FW from the screen here below, the .bin will now be a modded one with the low power consumpiton feature?

[image]

Note: I've checked the shasum between the first original .bin firmware and the second modded .bin file and I can see 2 different ashes, this would make me think that I may be right...

Please let me know if I am doing it correctly or if I am completely on the wrong way :) 

Thanks!

Deo

Sn0wfreezeDev, 2021-12-08:
Hi @Furtivo360, 

Please follow @acalatrava's Readme to compile the firmware and install it on your device. It does not work to move it into the OpenHaystack app since the installation is more involved here. 

Kind regards
Alexander

mearsy, 2022-03-04:
> Hi @Furtivo360,
> 
> Please follow @acalatrava's Readme to compile the firmware and install it on your device. It does not work to move it into the OpenHaystack app since the installation is more involved here.
> 
> Kind regards Alexander

I just bought one of these for testing

https://www.aliexpress.com/item/32826502025.html

My question is, @acalatrava provides precompiled NRF51 firmware at

https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative/compiled

So I moved it into the app as such Openhaystack --> Contents --> NRF.

NRF folder only seemed to contain firmware for NRF52 so I renamed the format on my file to NRF51_NRF51822.openHayStack.hex.

First thing I noted. His compiled firmware seemed big at 2.2mb vs the 500k for the existing ones

Second question. Do I have the naming format correct. i.e will Openhaystack detect the device as being a NRF51822 and pick up and use the new firmware.

Finally the most important newbie question. How to program it. Am I correct in assuming just connecting the STlink V2 up to the correct pins, I will be able to flash the device directly from the Openhaystack app with the option "Deploy NRF device"?

Apologies for the N00b questions.

[26 more replies]