seemoo-lab/openhaystack #58: Deploy for ESP32 with version 0.37 not working

Terrormixer3000, 2021-05-10:
**Describe the bug**
With the latest precompiled version (0.37) of OpenHayStack deploy to ESP32 does not work. After selecting the correct port for the ESP32, OpenHayStack directly shows me the error message: 
"**Could not deploy**
Deploying to microbit failed. Please reconnect the device over USB"

With version 0.36 everything works as expected.

**OpenHaystack version:**
0.37 (pre compiled)

**macOS version:**
11.3.1

**Additional context**
Add any other context about the problem here.

empicano, 2021-06-29:
Same for me on MacOS 11.4 and OpenHaystack 0.3.7, directly shows the error message for ESP32s

jasonc624, 2022-01-31:
Is there an update for this

Sn0wfreezeDev, 2022-01-31:
Could be related to: https://github.com/seemoo-lab/openhaystack/issues/66

mearsy, 2022-02-16:
I also have the same issue with a DF Robot fire beetle. I've updated the drivers, but I still get the failed to deploy too microbit. I tried it via flash_esp32.sh but I don't understand the messages at the end. It just doesn't seem to deploy. Tried it on build 0.36 and got the same issue.

It doesn't seem to be related to serial port drivers. Port is detected fine

Nicolas-Air-4:ESP32 nicola$ ./flash_esp32.sh -p /dev/cu.usbserial-1420 "il1JK/sN2tsmYLPPFlpBnP1iUGscKS30HaF+MQ=="
esptool.py v3.2
Found 3 serial ports
Serial port /dev/cu.usbserial-1420
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 94:3c:c6:07:d1:d4
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.1 seconds.
Staying in bootloader.
esptool.py v3.2
Found 3 serial ports
Serial port /dev/cu.usbserial-1420
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 94:3c:c6:07:d1:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000efff...
Flash will be erased from 0x00010000 to 0x000b3fff...
Compressed 25440 bytes to 15916...
Traceback (most recent call last):
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 5143, in <module>
    _main()
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 5136, in _main
    main()
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 4602, in main
    operation_func(esp, args)
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 3851, in write_flash
    blocks = esp.flash_defl_begin(uncsize, len(image), address)
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 154, in inner
    return func(*args, **kwargs)
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 909, in flash_defl_begin
    self.check_command("enter compressed flash mode", self.ESP_FLASH_DEFL_BEGIN, params, timeout=timeout)
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 468, in command
    p = self.read()
  File "/Users/nicola/esp-idf/esp-idf/ESP32/venv/bin/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration

MCKLtech, 2022-04-09:
> > esptool.py", line 5143, in
> > _main()
> 
> Think I found the solution. Dropping the baud rate to 115200 in flash_esp32.sh seemed to let it go through.

This also solved by the flash issue with D32 Pro

Line 18: BAUDRATE=115200