seemoo-lab/openhaystack #60: add time and date to past location pins

makiele, 2021-05-13:
A clear and concise description of what the problem is.

When using the feature to show all locations from the past 7 days, I would like to see at what time the location was recorded. At the moment, when click a red pin nothing happens. Therefore, it is impossible to tell when the tracker/tracked object was at the various locations in the last 7 days.

**Describe the solution you'd like**
While clicking a pin from the last 7 days, OpenHaystack.app should show time and date of the tracked location.

**Describe alternatives you've considered**
In addition, a possibility to recreate the path the tracker took in the past 7 days would be interesting. At the moment, I‘m using it to track my car. A feature which would show which routes the tracker took would be interesting

**Additional context**
Add any other context or screenshots about the feature request here.

thank you for your awesome work!

darkroasted, 2021-05-13:
+1 Would also like to see this feature

dererdmann, 2021-05-27:
+1 would be very helpful! Also raw data access to the time and location e.g. exported in a text file could be useful.

Pondiniii, 2022-04-09:
+1 I'm not the only one who had this idea :)