seemoo-lab/openhaystack #61: Any support for macOS lower than BigSur?

Steve-0628, 2021-05-14:
For some reasons, I don't have an Apple device with macOS BigSur. Is there any chance that this project will be available for macOS Catalina, for example?

I tried building myself, but I have no knowledge about Swift, so it was not possible for me. 

I do understand that the fastest solution is for me to get BigSur. 
I really appreciate if it is possible, thanks!

acalatrava, 2021-05-16:
It’s technically possible to make it work from MacOS Catalina. I’m trying to build a simple/command line version for it.

urbfigviera, 2021-05-16:
Pls I want to unlock all my devices locked

jimmymcheung, 2021-06-05:
As far as i seen in the find my on macOS 10.15.7 (19H1217), it does not have an option for object searching, so i assume it won’t receive any related data unless it comes as a whole bundle with find my device.