seemoo-lab/openhaystack #62: Play sounds (and possibly send arbitrary payloads to the tracked device)

AwesomestCode, 2021-05-24:
**Background**
AirTag and other official "Find My" compatible devices allow you to trigger a sound to be played. They also allow you to remotely turn on a feature called "lost mode."

**Request**
I would like to be able to trigger these action on OpenHaystack Tags (e.g., be able to press a button on the OpenHaystack app and send a signal to the OpenHaystack tag to initiate a playing a sound or triggering lost mode like normal AirTags).

**Additional thoughts**
* Perhaps the play sound and lost mode could be bound to pre-defined scripts as a way of triggering them and not necessarily to perform the functions Apple uses them for.
* If playing sounds and enabling lost mode are sent through a payload that can store arbitrary data, this could potentially allow for it to trigger more commands than just those two.
  * Failing that, it would be interesting if the lost mode and play sound signals could be modulated between in a binary code to send messages. Perhaps timing would make this infeasible.

Sn0wfreezeDev, 2021-05-25:
Thank you for the recommendation. 

The current firmware does not allow such features as it does not communicate at all with other devices using a Bluetooth connection. It only sends BLE advertisements and it is not connectable. Furthermore, most beacons / dev boards do not have a speaker to play sounds.

AwesomestCode, 2021-05-25:
> The current firmware does not allow such features as it does not communicate at all with other devices using a Bluetooth connection.
Yeah, I'm aware. I was just wondering if it was possible to implement.

hatomist, 2021-07-22:
I found [these tags](https://aliexpress.ru/item/1005002489678133.html) and discovered that thay feature a TLSR8266 or ST17H25 MCU, which has BLE onboard and a speaker as well. Futhermore, apparently it [can be programmed](https://github.com/pvvx/TlsrComProg) with a cheap USB-TTL adapter, what makes it even more appealing than an NRF51822 aliexpress beacon (depending on how long it will last from one battery). I'll take a closer look as quick as I'll get one.

solarkraft, 2021-10-01:
The ability to trigger arbitrary actions would be extremely valuable. How much data could theoretically be communicated?

Sn0wfreezeDev, 2021-10-04:
We are working on a new firmware, which might get new capabilities. 
For now we are concentrating on using less energy and rotating the Bluetooth identifiers

aaronnabil, 2021-10-04:
To try to minimize energy use I wrote an entirely standalone (no bluetooth library) version, it directly sets up the radio and sends a raw packet.  It is based on a Nordic demo (I think it was called something like "solar beacon").  I can post some charts of the energy usage of mine.

lennyerik, 2021-12-26:
I believe that the Apple Offline Finding protocol is one-way only, since none of the features provided by Apple need two-way communication. Playing a sound from an AirTag is only possible when the user is in close proximity to the tag (a friend of mine who owns an AirTag confirmed this). This means that the user's device probably just connects to the device via BLE (not via the OF network!).

"Lost mode" is probably even simpler: the device enabling lost mode would only need to communicate with Apple's servers. Afterwards the website encoded on the NFC tag of the AirTag is updated to display the contact information and your Apple devices are notified when the AirTag is found (both of which is handled by Apple's servers internally). The advertisement messages from the AirTag are sent out regardless of whether lost mode is on or off.

So, yes playing a sound and sending arbitrary data to the tag is possible, but probably only in BLE range of the tag and not actually over long distances using the the Offline Finding network (since there is no bidirectional communication as far as I know).