seemoo-lab/openhaystack #66: Could not deploy on ESP32 (ESPRESSIF)

KatzeMau, 2021-06-07:
When i want to deploy an Airtag i get this error: Deploying to microbit failed. Please reconnect the device over USB

(To be honest, I don't know what to add here, since I have little affection for the mateire)

**OpenHaystack version:**
Version 0.3.7

**macOS version:**
11.4 (20F71)

[image]

[image]

lanma, 2021-06-10:
I got the same error message.
macOS version:
11.2.3
M1

mowtschan, 2021-06-10:
I guess, I "know" the problem and maybe there is a workaround for that! 
You can try to modify a little bit the bash script under /Applications/OpenHaystack.app/Contents/Resources/ESP32/flash_esp32.sh:

instead of:
```
esptool.py --after no_reset --port "$PORT" \
    erase_region 0x9000 0x5000
esptool.py --before no_reset --baud $BAUDRATE --port "$PORT" \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
```

try this:
```
esptool.py --baud $BAUDRATE --port "$PORT" \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
sleep 2
esptool.py --port "$PORT" erase_region 0x9000 0x5000
```

Then try again to flash your esp32 device with OpenHaystack app!

More background information for the "workaround" here: https://github.com/seemoo-lab/openhaystack/issues/43

KatzeMau, 2021-06-10:
> I guess, I "know" the problem and maybe there is a workaround for that!
> You can try to modify a little bit the bash script under [...]

Sadly it does not work, via the app
I used terminal, but i get error Failed to write to target RAM (result was 01070000)

`mac@MacBook-Pro ~/Downloads (master)> sh /Users/mac/Desktop/OpenHaystack.app/Contents/Resources/ESP32/flash_esp32.sh -p /dev/cu.usbmodem14201 -s w96MLz61+gpc9XGRFcRMxmSZBBRe9BXryG8zLAxqLg4=
esptool.py v3.1
Serial port /dev/cu.usbmodem14201
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:12:48:2c
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
cleanup ...
mac@MacBook-Pro ~/Downloads (master) [2]>
`

alaouiali, 2021-06-19:
I have the same issue. so if I flash with using the "advertisement key" and use in the command below. Would that work ? not sure if advertisement key is the public key. 
./flash_esp32.sh -p /dev/yourSerialPort "advertisement-key"

[image]

Sn0wfreezeDev, 2021-06-21:
I'll definitely have a look into this issue soon, but I have a lot of other obligations at the moment. 
The advertisement key is the public key that will be published. The public key itself has some additional bytes at the start for standard compliance reasons

KatzeMau, 2021-06-21:
> I'll definitely have a look into this issue soon, but I have a lot of other obligations at the moment.
> The advertisement key is the public key that will be published. The public key itself has some additional bytes at the start for standard compliance reasons

Thank you!
Sadly @alaouiali method did not work :/

But what causes the error ?
The key is to long for the standard compiler ?

KatzeMau, 2021-06-29:
@Sn0wfreezeDev Any update so far?

KatzeMau, 2021-07-25:
Will this be fixed or not ????
@Sn0wfreezeDev @schmittner

Sn0wfreezeDev, 2021-07-28:
When the schedule allows it, it will be fixed

shivharis, 2021-11-13:
@Sn0wfreezeDev Not sure why you closed this without specifying a reason for closing. Someone else could have picked it up and attempted to fix this situation.

KatzeMau, 2021-11-13:
I found a kinda solution
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-904390716
I Downloaded the driver and flashed the ESP without problem :) 

https://www.virustotal.com/gui/file/cb04c089037319d739e767a15f8a4cc45d63495e
[CH9102_Mac_Driver.zip](https://github.com/seemoo-lab/openhaystack/files/7531890/CH9102_Mac_Driver.zip)
74812cfe5818639c9c5abd40
https://www.virustotal.com/gui/file/f171254e533eafed4e9eabe5ef23520d31ceecc057a4b7d1855907182a252720

timbodeit, 2021-11-13:
@shivharis Read my comment above. It is an issue. But it is not an issue with openhaystack.

timbodeit, 2021-11-13:
@KatzeMau As referred to above: Using the driver mentioned in the upstream issue on the esptool project seems to work.

The potential security implications of installing drivers of unconfirmed origin is something everyone has to decide for themselves.

lanrat, 2022-06-25:
I was able to manually flash my ESP32 by editing the last command in `flash_esp32.h` and removing the `--before no_reset` argument. (only from the `write_flash` command`)

So the end result is:
```bash
esptool.py --baud $BAUDRATE --port "$PORT" \
    write_flash 0x1000  "$SCRIPT_DIR/build/bootloader/bootloader.bin" \
                0x8000  "$SCRIPT_DIR/build/partition_table/partition-table.bin" \
                0xe000  "$KEYFILE" \
                0x10000 "$SCRIPT_DIR/build/openhaystack.bin"
```