seemoo-lab/openhaystack #68: Apple Mail is Crashing when OpenHaystack App is active

DrSchottky, 2021-06-29:
**Describe the bug**
After enabling plugin, If I launch OpenHaystack App with Mail opened (or vice versa) Mail crashes

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Mail
2. Launch OpenHaystack

**Expected behavior**
Mail not crashing

**Screenshots**
[image]

**OpenHaystack version:**
OpenHaystack Version 0.3.7 (1)

**macOS version:**
Mail Version 14.0 (3654.100.0.2.22)
macOS Version 11.4 (20F71) on a MacBook Pro (15-inch, Mid 2012)

ImSukh, 2021-10-17:
Not sure if you're still looking into this but I think this is caused by not having an iCloud email. Do you have a iCloud email account setup with Apple Mail?

etmatrix, 2022-03-05:
@Sn0wfreezeDev so iCloud Mail is mandatory? Is there a way to bypass it?

Inf3rnus187, 2022-10-19:
Hi.

I was trying to use OpenHayStack on m1 MacBook Pro.

I am facing the same issue.

MacOS 12.2
Mail 15

My icloud use another e-mail provider.

Is that necessary to use icloud mail ?

I havé tried to compile the app and the mail bundle. The app run but the bundle Say not good uuid for mail.

If i use precompiled binary of the bundle mail crash like in this Topic.

Any suggestion maybe ?

I would like to say thanks a lot for you Work !