seemoo-lab/openhaystack #70: How to track the location of Android phone?

jwli6, 2021-07-01:
I have an Android phone. How to add it to openhaystack？

hatomist, 2021-07-05:
Hello!
I've tried to do that using [EFR Connect](https://github.com/SiliconLabs/EFRConnect-android) but then I realized that I need to set a custom BLE device address, which's not supported by the Android BLE framework.
So I don't think it would be possible to find your Android phone using Find My network, at least using stock firmware.

Well, you still can attach an Aliexpress NRF beacon to it somehow :sweat_smile:

Sn0wfreezeDev, 2021-07-07:
Yes @hatomist is completely right. 
You will get it to work on rooted Android, but we never tried that. 
One option for this would be to use another tool of us called Internal Blue: https://github.com/seemoo-lab/internalblue 
It does not run on all Android phones

closed 2021-07-07 by Sn0wfreezeDev