seemoo-lab/openhaystack #71: What do "datePublished" timestamps exactly represent?

zakyum, 2021-07-05:
First off, thank you for this amazing reverse engineering project (!) and for opening one of apple's platforms to the (hacker) people 🤩. Really appreciate being able to build my own AirTag and make use of my location data!

Going to the question, in your paper "Who Can _Find My_ Devices?" in section "6.3 Finding" you talk about how each report has a timestamp associated with it. Reading further, I got the impression that the timestamp represents the time when the report was generated, i.e. when the OF advertisements was picked up by the finder. But after downloading the reports and plotting a simple path based on (increasing) time, I noticed that sometimes a report's time is later than expected. I.e. the report's reported location accuracy is high and fits with the actual location I passed by, but it's "datePublished" is late by several tens of minutes when compared with the actual time I passed by that location. Also, I saw that sometimes several reports' "datePublished" time is identical but their location different.

Because of that I wanted to ask what the "datePublished" timestamps exactly represent. Judging from what you wrote in your paper, I'd say it's the time when the report was generated, but looking at the data I'd say it's the time the report was uploaded.

Hope you can clarify that :)

Sn0wfreezeDev, 2021-07-07:
That's actually a good question! Thank you very much for reading the paper. 
So the easy answer to this is: There are two timestamps. The timestamp that reports the time when the report has been received and the `datePublished` which is the date when report has been uploaded to Apple. 

The `datePublished` is part of the JSON, while the actual `timestamp` is part of the location report binary. 
You can see the difference in our code here: https://github.com/seemoo-lab/openhaystack/blob/acdae59b3985be18c054c2140d688582646d7fc7/CVE-2020-9986/OFReadKeys/OFFetchReports/FindMy/Models.swift#L57 

```swift
var timestampData = payload.subdata(in: 0..<4)
    let timestamp: Int32 = withUnsafeBytes(of: &timestampData) { (pointer) -> Int32 in
    // Convert the endianness
    pointer.load(as: Int32.self).bigEndian
}

// It's a cocoa time stamp (counting from 2001)
self.timestamp = Date(timeIntervalSinceReferenceDate: TimeInterval(timestamp))
```

closed 2021-07-07 by Sn0wfreezeDev

zakyum, 2021-07-08:
Thank you, now it makes sense! And thanks for pointing me to the relevant code, totally overlooked that...