seemoo-lab/openhaystack #72: The reported location is not accurate

iamfaith, 2021-07-06:
**Describe the bug**
I try on Raspberry pi and ESP 32. Both of them have reported the wrong location comparing with the location in Find my app  from my iPhone.

**Expected behavior**
Showing the same location as shown in Find my app

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenHaystack version:**
Version 0.3.7 (1)

**macOS version:**
Big Sur

**Additional context**
Add any other context about the problem here.

zakyum, 2021-07-06:
Yeah, that's to be expected. The Find My app preprocesses/combines raw location reports to give a better approximate of the true location. Meanwhile (as far as I got from the code) the OpenHaystack app only shows the last raw location, i. doesn't do any extra preprocessing/combing. You'd have to do that yourself.

iamfaith, 2021-07-07:
> Yeah, that's to be expected. The Find My app preprocesses/combines raw location reports to give a better approximate of the true location. Meanwhile (as far as I got from the code) the OpenHaystack [...]

Thanks for your reply. Could you please give me some ideas about how to do extra preprocessing/combing? Thanks

Sn0wfreezeDev, 2021-07-07:
Hi, 
yes the current OpenHaystack does not aggregate last locations. 
Also the Find My app has two different modes: 
A) The location is fetched directly from your devices without offline finding 
B) One of your devices is offline (no internet connection) and will be found by nearby devices. Those nearby devices will report their location. 

OpenHaystack only works with B and uses the Find My network. This will always be less accurate, as the devices that find an OpenHaystack accessory will use their own location. This location can be up to 100m away from the actual device. 
To get better results we could aggregate several reports, but we decided to just show the last report and allow you to view past reports by clicking at the clock icon. 

So this is not really a bug, but more part of the design of the Find My network

closed 2021-07-07 by Sn0wfreezeDev

iamfaith, 2021-07-07:
> Hi,
> yes the current OpenHaystack does not aggregate last locations.
> Also the Find My app has two different modes:
> A) The location is fetched directly from your devices without offline finding
> B) One [...]

Dear Sir, Thanks for your explanation. I understand this. But what I try is to put my ESP32 devices near my iPhone in this case that my iPhone should report the accurate location for my ESP32 devices. While I check the reported location in Openhaystack is about 2 kilometers away from the Find my app's locations. 

Do you mean the reported location is reported by other devices instead of my iPhone?

Sn0wfreezeDev, 2021-07-08:
Yes. Any iPhone could report your locations. 
Also it can take up to 3 hours before an iPhone reports a location so you can see it in OpenHaystack. 
But the average for updated reports is around 26min.

iamfaith, 2021-07-08:
> Yes. Any iPhone could report your locations.
> Also it can take up to 3 hours before an iPhone reports a location so you can see it in OpenHaystack.
> But the average for updated reports is around 26min.

Thanks. I know and I have already seen the reported location. But it's far away from my iPhone, about serval kilometers. I am sure there is only my iOS device nearby. Is it a bug for the location? I mean the reported location is not accurate.

zakyum, 2021-07-08:
You might check the "accuracy" field of the corresponding location report to see what you get. A dumb but good "filter" which I use is to remove all reports with an accuracy value of >160 (greater means less accurate). Should remove most of the noisy data, while keeping everything else.

Tho never had reports off by several kilometers 😅. Are you maybe living in proximity to a train station or highway?

Sn0wfreezeDev, 2021-07-09:
I only had really off reports when next to a street and the iPhones that received the signal might have got their location fix a lot later

FChong96, 2022-06-15:
It seems I have the same problem.