seemoo-lab/openhaystack #75: OpenHaystackMail.mailbundle incompatible with macOS Big Sur Version 11.5.1

sebi5361, 2021-07-28:
Since I updated to macOS Big Sur **version 11.5.1**, I get this error message:
[image]
...in french, which translates to:

> **Incompatible Modules Disabled**
> Mail has disabled the following modules:
>
> OpenHaystackMail.mailbundle
>
> Contact the editors of these modules to
> obtain versions compatible with Mail 14.0.

...and thus OpenHaystack **version 0.3.7 (1)** doesn't work.

sebi5361, 2021-07-28:
It seems pull request [**Added a bunch of UUID's into the plist for Mail app version 14**](https://github.com/seemoo-lab/openhaystack/pull/64) fixes this.
Will try that...

sebi5361, 2021-07-28:
Indeed after compiling (using Xcode) [VladutLP's OpenHaystack fork](https://github.com/VladutLP/openhaystack) and running it, I can confirm it works again with Mail version 14.0 and macOS Big Sur version 11.5.1.

Sn0wfreezeDev, 2021-08-06:
Thanks for the pull request and the fork. 
And sorry for the delays. 
I just added the relevant code for compatibility with 11.5

closed 2021-08-06 by Sn0wfreezeDev