seemoo-lab/openhaystack #77: Is this legal enough for businesses to use it?

HakuG, 2021-08-05:
Is OpenHaystack legal enough to build your own Tile system or software over it? I know it has some AltMail and AltStore dependencies, which Apple I believe does not support, but I could see many industries wanting to play over Apple's Find My Network.

sebi5361, 2021-08-07:
Interesting question... I am wondering too about Apple's feeling about people using their Offline Finding _Find My_ network?
From my understanding every one is allowed to emit a BLE signal even if it complies with the AirTag standards and is registered on the _Find My_ network. And any Apple product owner is allowed to access the _Find My_ servers. It seems that, by design, you can withdraw any position log you want (indexed using the public key?). The rest is history: once you have the log and you know the private key, you can decrypt the position data. Why isn't Apple forcing you to register your Apple certified beacon (AirTag) and let you to withdraw all the logs is certainly due to the fact that on AirTags public-private key pairs are rolling (to be confirmed, contrary to OpenHaystack beacons) which makes it impossible to register all the public keys for a given Apple certified beacon. I am not sure this latest statement is true and there is no way for Apple to have you register a beacon and withdraw its data only. From a legal standpoint, I am not sure Apple would let you use their Offline Finding network for a commercial use though. The paper written by the authors gives some more hints.

zefoo, 2021-08-08:
Appreciate the question. But the answer may be more important to your own business ethics and/or a legal team’s answer. Not intending to troll here, but I suspect the answer - around legalities - is, “it depends.” 

If I had to guess, technically speaking, the answer is probably no… haha.

Although, I’m a bit confused - are you inferring that openhaystack can tap into Apple’s AirTag system? Unless I missed something, that’s not possible due to the way Apple AirTags work… unless you’re strictly speaking to the “Find My” equivalent - in which I believe openhaystack and pyicloud both tap into that (but, last I checked, airtags is it’s own monster that is technically unable to integrate with… and as such, an iOS device is required to set up / pair / track). Would be curious to “fact check” my understanding here if someone can confirm I’m tracking appropriately here.

Sn0wfreezeDev, 2021-08-23:
My feeling tells me that a commercial system based on this is illegal to Apple's terms and conditions. I don't if they can fine you legally for this (I am not a lawyer) but they will probably try to (and they have expensive lawyers). 
That's why I think Apple will try to close the access to their server in the long run. 

But as @sebi5361 said: It's technically impossible to pre-register all keys available. What is possible is to restrict Apple's operating system even more. So that's why they will remove Apple Mail Plug-Ins in macOS 12.x or 13.x at latest. 
They clearly said that in a WWDC session. 

So OpenHaystack can probably only exist, because it runs as an open-source  project without any profits.

closed 2021-08-23 by Sn0wfreezeDev