seemoo-lab/openhaystack #78: Mail Incompatible Plug-ins Disabled Issue

am-rutherford, 2021-08-06:
Following the installation steps and getting this warning on step 5 and can't seem to get around it, any ideas on how to fix this?

[image]

am-rutherford, 2021-08-06:
Downloaded the source files instead of the pre compiled version and built it using Xcode to fix this

closed 2021-08-06 by am-rutherford

Sn0wfreezeDev, 2021-08-06:
I just updated the source code for it today. Also a new release is available now: 
https://github.com/seemoo-lab/openhaystack/releases/tag/v0.3.8 

Thanks to a pull request that has been open for a while: https://github.com/seemoo-lab/openhaystack/pull/64