seemoo-lab/openhaystack #79: Unable to allow Apple Mail Plugin to activate.

rolledankle1000, 2021-08-16:
Hello Community. I found an issue when trying to run openhaystack. So I create a new user, and download the application. It then asks me to install the mail plugin, which I install to its pre-determined folder, mail/bundles. I then run the code in terminal to see the plugin in the preferences (If i do not run the terminal code, I see nothing, as expected). I then click on manage preferences, and I see the openhaystack.mail plugin. I click the check box, then hit allow the plugin. I then get the error, "Incompatible plugins disabled, mail has disabled the following plugin, contact the creator for versions compatible with mail 14.0". And then the mail application restarts with the plugin disabled. OpenHaystack opens, but the green dot in the corner does not appear, as it is not in connection with the FindMyNetwork. etc etc. Any way to get around this?

rolledankle1000, 2021-08-17:
Is this compatible with the latest OS from apple, i believe it is 15.5 Big Sur? or does it need to be a previous version to work?

Sn0wfreezeDev, 2021-08-23:
Hi @rolledankle1000, 

The newest version of the app should be compatible 11.5.1 
Now I see that there is 11.5.2 available, but I have not update to this one yet. Which one do you have installed? 15.5 is not a valid macOS version 😅

desq42, 2021-09-15:
Same here, 11.6…

Sn0wfreezeDev, 2021-09-15:
Likely there's a new UUID for macOS 11.6. 
I'll try to find out which one it is and update

Sn0wfreezeDev, 2021-09-22:
A new release with the correct UUIDs has been published yesterday. It includes all UUIDs necessary to run the plug-in in 11.6. 
Unfortunately the update process might show that it failed to update the plug-in even though it worked.

closed 2021-09-22 by Sn0wfreezeDev