seemoo-lab/openhaystack #80: Mail Plugin incompatible with Mail app v15.0 from MacOS Monterey

Astrr, 2021-08-31:
**Describe the bug**
Trying to enable the mail plugin results in the error message about it not being compatible with mail app v15.0

**To Reproduce**
Steps to reproduce the behavior:
1. Install MacOS Monterey Beta
2. Follow openhaystack installation instructions
3. Get the error at the plugin activation step

**Expected behavior**
The plugin being enabled.

**Screenshots**
[image]

**OpenHaystack version:**
Version 0.3.8 (1)

**macOS version:**
12.0 Beta

**Additional context**
It seems that the Mail app is switching to a new plugin format, which might be the problem

Astrr, 2021-08-31:
Fixed by https://github.com/seemoo-lab/openhaystack/pull/64#issuecomment-886591178

Sn0wfreezeDev, 2021-08-31:
@Astrr Can you confirm that the app still works with Monterey? 
I don't have a Mac with Monterey installed and I know that Apple wants to disable Mail Plug-Ins completely in the near future.

Sn0wfreezeDev, 2021-09-22:
I just updated the app with the newest UUIDs for 12.0 and 11.6 
The update mechanism for the plug-in works, but it is a bit rough at the moment. I'll look into this

closed 2021-09-22 by Sn0wfreezeDev

solarkraft, 2021-10-01:
Would be cool if this made it into a release, as I run into compliation errors.

Sn0wfreezeDev, 2021-10-04:
It is in the latest release. You can just download version 0.3.9

solarkraft, 2021-10-04:
Welp, then we have another bug on our hands, because it doesn't work. Has anything changed with the recent Betas (7/8)?

Sn0wfreezeDev, 2021-10-05:
Could be that there are changes in the lastest betas. Which version are you running? 
Could you check your Mail Plug-In UUID? 
That can be found by going to Applications→ Right Click on Mail → Select show package contents → Content → Open Info.plist in Xcode or any text editor. There is an entry called: `PluginCompatibilityUUID` please paste the UUID String here. 
It is the same for all macOS users. So it doesn't contain private data

solarkraft, 2021-10-05:
I'm on Monterey Beta 8 with OpenHaystack release 0.3.9. The PluginCompatibilityUUID is 224E7F96-2099-499C-A501-63FB68C79CD2.