seemoo-lab/openhaystack #88: Incompatible Plug-ins Disabled after laters OSx Monterey update 12.1 (21C52)

Furtivo360, 2021-12-18:
Hi,

OpenHaystack is not working after latest update of OS Monterey 12.1 (21C52)

```
Incompatible Plug-ins Disabled
Mail 
Mail has disabled the following plug-ins:
OpenHaystackMail.mailbundle
Contact the makers of these plug-ins for versions that are compatible with Mail 15.0.
```

Everything used to work fine before this latest update.

I've tried to uninstall the plugin, restart my Mac and follow these steps again:

- Download a precompiled binary release from our GitHub page.
- Alternative: build the application from source via Xcode.
- Open OpenHaystack. This will ask you to install the Mail plugin in ~/Library/Mail/Bundle.
- Open a terminal and run sudo spctl --master-disable, which will disable Gatekeeper and allow our Apple Mail plugin to run.
-  Open Apple Mail. Go to Preferences → General → Manage Plug-Ins... and activate the checkbox next to OpenHaystackMail.mailbundle.
- If the Manage Plug-Ins... button does not appear. Run this command in terminal sudo defaults write "/Library/Preference/com.apple.mail" EnableBundles 1
- Allow access and restart Mail.
- Open a terminal and enter sudo spctl --master-enable, which will enable Gatekeeper again.

But still not working.

**Screenshots**

[image]

[image]

**OpenHaystack version:**
Version 0.4.1 (1)

**macOS version:**
12.1 (21C52)

**Additional context**
Plugin used to work perfectly before update.

Thanks

Furtivo360, 2021-12-23:
Forgot to add --> Mail Version 15.0 (3693.40.0.1.81)

jess-sys, 2021-12-23:
Can reproduce.
Maybe try fixing the Info.plist inside (follow https://www.reddit.com/r/AltStore/comments/nvfu2b/alt_store_incompatible_plugins_disabled_fix_macos/)

Furtivo360, 2021-12-23:
Hi @jess-sys, thanks for your suggestion.
I've followed the guide but still getting the same error 

"Incompatible Plug-ins Disabled
Mail 
Mail has disabled the following plug-ins:
OpenHaystackMail.mailbundle
Contact the makers of these plug-ins for versions that are compatible with Mail 15.0."

Have you updated your mac already? Can you check on wich mail version you are on please?

Thanks :)

jess-sys, 2021-12-23:
Hi @Furtivo360,

I've got the exact same versions you posted.

I successfully got it to work after installing AltStore's mailbundle. Installing it unlocked the "Manage plug-ins" button inside Mail Preferences, then I was able to enable the previously loaded OpenHaystack.mailbundle.

Steps to reproduce:
* Install OpenHaystack latest release
* Follow the start-up instructions (allow the app to save the mailbundle inside ~/Library/Mail/Bundles)
* Install AltStore
* Go to the top status bar, click on "Install Mail Plug-in"
* Restart your Mail app
* Go to Mail > Preferences
* On the General tab, you now have a button "manage plugins" on the bottom left corner

buzali, 2021-12-23:
@Furtivo360 I was able to install the plug-in again by following these instructions https://github.com/rileytestut/AltStore/issues/905#issuecomment-985985323

Furtivo360, 2022-01-03:
Still not able to use the plug-in...I don't understand :(

breakingcraft, 2022-01-20:
I was able to reproduce the issue after OS update and I solved the issue with downloading the newest release and install it.

Furtivo360, 2022-01-24:
Yes! Can confirm that with the latest OpenHaystack verison the issue got solved! 👍 
Thanks!

closed 2022-01-24 by Furtivo360