seemoo-lab/openhaystack #90: what is the adv pub key

kolet, 2021-12-24:
hi there,  i am pretty new to all of this , 

i've read lots of issues and comment on this repo just to understand a lilbit more .
something i could not figure out is .

1, how does the pub key looks like ? can some one perhaps shows how does that looks and how to implement it ?

2, since the README says that iphone users cannot distinguish an openhaystack devices (a beacon with openhaystack flash on it)  and a real air tag ,  does the iphone users receives a notification aswell if an unknown beacon is nearby for to long ?

from what i remember about 6-7 month ago it was 3 days untill the iphone user gets a notification .

can some one clarify those 2 questions? thanks 

(btw if there is a guide on how to flash the firmware + the adv key etc.. via a video that would be awasome .)

Sn0wfreezeDev, 2022-01-03:
1: The implementation details are shown here: https://github.com/seemoo-lab/openhaystack/tree/main/Firmware 
A public key is just 28 bytes long data. So there is not much to see. 
2: Yes since iOS 15 iPhone users will also get tracking alerts for OpenHaystack tags. It took some time for me to get a notification in my test 

I actually made some videos, but I never published them 😄. I just uploaded one to YouTube, you can find it here: https://youtu.be/TTSAcBQyVh0

kolet, 2022-01-03:
@Sn0wfreezeDev  cool i just saw your vid ! please post more !
but it does not show how to flash it etc.. :)

i'd hoped to see a full flashing guide with entering the pub key and making it work etc..  a-z kinda stuff .

another question .

how long did it take for you to get a notification ?

also is that possible to disable that feature ?  if my bag/bike/laptop is getting stole i dont want the thief to know i am tracking my device .

Sn0wfreezeDev, 2022-01-03:
Disabling tracking notifications is not a good option. Since they are there for a good reason. They also cannot be disabled at the moment. But it takes a while until you get them. I used it with my car and it took a couple of miles and arriving at home to get a notification. Apple only sends them when you arrive at home normally

kolet, 2022-01-10:
" I used it with my car and it took a couple of miles and arriving at home to get a notification. Apple only sends them when you arrive at home normally" 

what do you mean , can you elaborate ?

Sn0wfreezeDev, 2022-01-10:
No I cannot describe it more clearly, that's just how I remember it. Apple seems to have invested more time in making the tracking detection more robust and I think this is generally a good thing to protect against abuse

kolet, 2022-01-14:
on the other hand , if your stuff got stole , and you did not notice it in time or you are out of town  or you are not able to get there because it become far from you .
and alot more reasons as why you want to disable it or at least extend the time .

the person that stole your shit can and will receive a notification after a period of time (i am not sure how long 1 day?) 
upon receiving the notification he will search for the tag , remove it or destroy it and your stuff is gone forever .

if a person wants to track some one or a car idk  they are some "spying" gadget thingy online that cost around 30$ and will last for month powered on with a gps on it.

what i did not understand is this phrase  "Apple only sends them when you arrive at home normally"  ( i dont own an airtag so idk how it act/works)

Sn0wfreezeDev, 2022-01-14:
I think in many cases, you will have the location of your stolen items before the thief has removed the tag and since the locations are stored for a week you can always see the location path. 

Since we also have a project that tries to inform Android users about tracking devices, I think the being stalked issue weighs more than the stolen items issue. 
https://github.com/seemoo-lab/AirGuard