seemoo-lab/openhaystack #93: The Microbit firmware mention a Public Key but doesn't say where to get it.

maeln, 2022-02-04:
Hi,

The README.md for the Microbit Firmware mention the following:

> You need to specify a public key in the firmware image. You can either directly do so in the [source](https://github.com/seemoo-lab/openhaystack/blob/main/Firmware/Microbit_v1/offline-finding/main.c) (public_key) or patch the string OFFLINEFINDINGPUBLICKEYHERE! in the final firmware image.

It is not explained where and how to get this public key.
The openhaystack app give access to the key ID, advertisement key and the symetric key. But none seem to fit the 28 characters of the original string in the firmware source.

Is it possible to clarify which pub key need to be put there ?

Thank you

Furtivo360, 2022-02-04:
Hi @maeln! 

I've been through the same issue but at the end I’ve been able to compile my Low Battery FW.bin.

You need to copy the Base64 advertisement key and generate a HEX string.
[image]

after that, you can use this bash script (thansk H. for this 👍) from terminal to generate the hexadeimal value of your key:

```
ADV_KEY_BASE64="base64 advertisement key from openhaystack"
NEW_ADV_KEY_HEX=$(base64 -d <<< $ADV_KEY_BASE64 | xxd -p)

STRINGREPLACE=$(echo -n 'OFFLINEFINDINGPUBLICKEYHERE!' | xxd -p)

echo find this $STRINGREPLACE

echo replace it with $NEW_ADV_KEY_HEX

```

Now you will need a HEX viewer (I.e. Hex Fiend)

From the HEX program, open the FW.bin and search for the string (from the script “find this xxxxxx”) and replace it with the new hexadecimal value always from the script (“replace it with xxxxx”)

Save with the name that you like and you’ve done it!

You can now flash that new .bin :)

Hope it helps.

D

maeln, 2022-02-07:
@Furtivo360 Thanks!

closed 2022-02-07 by maeln