seemoo-lab/openhaystack #94: Purchase NRF51822 from amazon

EzPressPlus, 2022-02-06:
So I'm thinking about ordering an NRF51822 from amazon but one thing I'm not sure about is how will I connect this to my computer to deploy from OpenhayStack Application on my mac? This is the one I'm planning on purchasing. 
I got the application on my mac and installed the Mail Plugin just fine. What else will I need to connect the computer to NRF51822
thank you. 

https://www.amazon.com/Comimark-NRF51822-Wireless-Bluetooth-iBeacon/dp/B07XY51FYQ/ref=pd_sbs_1/130-4093599-8225501?pd_rd_w=Xfann&pf_rd_p=cd718a0c-f7e0-41b6-9f23-6496d85d1998&pf_rd_r=Q69RVG9PYBJGEMS2M7C1&pd_rd_r=95a4efb8-d251-4cc8-a912-b59e0d0a3b24&pd_rd_wg=WdCYk&pd_rd_i=B07XY51FYQ&psc=1

mowtschan, 2022-02-06:
Look at the issue #35 
@sthorley wrote a great guide https://github.com/seemoo-lab/openhaystack/issues/35#issuecomment-828338751
Also you will need https://www.ebay.de/itm/255283221952?hash=item3b7010d5c0:g:Uq0AAOSwYRJdQWxj

iceblu3710, 2022-02-23:
I am looking at these all in one units and not sure which will work: 

[Aliexpress Link](https://www.aliexpress.com/item/32954025507.html?spm=a2g0o.search0304.0.0.4fcd5c49qL2t3j&algo_pvid=91a268f2-a291-46b3-8631-be91f2f0036c&algo_exp_id=91a268f2-a291-46b3-8631-be91f2f0036c-0)

I plan on simply tracking if an asset is inside the building or not. Beacon rate can be every 4 hours for the longest battery life.