seemoo-lab/openhaystack #95: Missing firmware source for the NEW NRF code - rolling keys.

TylerHallPureEngineering, 2022-02-11:
In the repo, there are obvious firmware links to:
- esp
- linux_hci
- microbit

for microbit -> the firmware given in that directory is using the fixed public key at compile time:
https://github.com/seemoo-lab/openhaystack/blob/b65a6e6be097faa836237bd5405f9d8f1d81d2c9/Firmware/Microbit_v1/offline-finding/main.c#L41

There is NEW NRF supported firmware that includes symmetric keys and generating a new rolling public key:
https://github.com/seemoo-lab/openhaystack/blob/main/OpenHaystack/OpenHaystack/HaystackApp/Firmwares/NRF/flash_nrf.py

This is using static generated hex files:
NRF52_NRF52832_openHayStack.hex
NRF52_NRF52840_openHayStack.hex

I would expect to find some firmware that has the variable swap much like in Microbit_v1/offline-finding/main.c#L41
But for the symmetric keys:
OFFLINEFINDINGSYMMETRICKEYHERE!

The NRF code rolling keys support was included in the app here:
https://github.com/seemoo-lab/openhaystack/commit/278fe4e30d17d4802724112b97287cb0ea07b7c8
and additionally prompt screen here:
https://github.com/seemoo-lab/openhaystack/commit/005d642dd8d64eaed1aae0f46e2588c5653184e9

My question,
where is the source code used to generate NRF52_NRF528xx_openHayStack.hex?
Would love to take a peek, potentially to match on other platforms. Considering the rest of the firmware has been posted, I figured that I could find this firmware source as well.

Thanks for all the good work! Super impressive.

bencorrado, 2022-11-01:
Hey @NeoZ16 is this something you could help with?  I would love to add some additional functionality to the nRF52 firmware!  Thanks!

beckmx, 2023-05-14:
Hello guys, I also noticed the same pieces of code, I received an nrf52810 and was wondering how you guys compiled those firmwares.

Greetings!