seemoo-lab/openhaystack #97: Can Openhaystack be used to track locations of „my“ AirTags (the original ones from Apple)?

ctschach, 2022-02-13:
Can Openhaystack be used to continually track the location of my AirTags? No, not for stalking. I would just love to record my AirTags positions over time and add them to a database…

Itheras, 2022-02-17:
Everything is possible with enough effort but I am inclined to say not really in this case without some problematic hurdles to overcome.

luke-jr, 2022-05-12:
What if you don't care about pairing with Apple stuff, and only want to use openhaystack?

swansonba, 2022-05-19:
This is a great question for folks that don't have an iOS device but would still like to use AirTags.  (Because the hardware is pretty good!)

brianr2600, 2022-06-20:
The airtag keys are periodically written into a plan local file by the Mac OS findmy app - since running the proxy already requires a Mac, it shouldn't be too high of a hurdle. See section 10 in Heinrich, Alexander, Stute, Milan, Kornhuber, Tim and Hollick, Matthias. "Who Can Find My Devices? Security and Privacy of Apple’s Crowd-Sourced Bluetooth Location Tracking System" Proceedings on Privacy Enhancing Technologies, vol.2021, no.3, 2021, pp.227-245. https://arxiv.org/pdf/2103.02282.pdf

luke-jr, 2022-06-20:
@brianr2600 The mentioned "Keys" subdirectory doesn't appear to exist in Big Sur at least.

luke-jr, 2022-06-20:
#37 mentions a "MasterBeacons" directory, which does exist with files for each airtag, but I don't really have time to investigate if there's a way to use those.

Sn0wfreezeDev, 2022-06-21:
The method of @brianr2600 would work the best. It requires you to disable some security and sandboxing features, but these files exist (as mentioned in our paper). But the issue is that Apple changes its format regularly. So at the moment, we don't have a working parser. The last parser that we created was working for macOS 10.15.4

advename, 2023-07-11:
Any updates if it's possible to track genuine AirTags with OH?
Its use case has been questioned in similar posts and to be quite frank, not everyone is skilled enough to build their own. 
AirTag's are ready-made trackers, but not everyone has access to an iOS or Mac device.

Sn0wfreezeDev, 2023-07-19:
Hi @advename , 
Tracking AirTags with your OpenHaystack needs way more skills than building your own trackers. 
That's very complicated to achieve.

maz3max, 2024-11-04:
This project manages to pull it off: https://github.com/malmeloo/FindMy.py
It actually works! You just need a mac or mac VM (https://github.com/Coopydood/ultimate-macOS-KVM) to get the OwnedBeacons plist files. Note that you should never give those to anyone, they can secretly track you with that.