seemoo-lab/openhaystack #98: Microbit being seen a lot more than ESP32

mearsy, 2022-02-16:
Odd question. I have both a microbit v1 and and ESP32-WROOM both with the firmware from the current build. They are sitting side by side in the same location in my car on the drive. In the past hour the microbit has been seen 10 times. The ESP32 only twice. Obviously for tracking this is less than ideal. I am wondering why this might be the case?

Itheras, 2022-02-17:
Having played with many different microcontrollers I suspect the difference in signal strength and quality between them as the most probable factor.

mearsy, 2022-02-17:
> Having played with many different microcontrollers I suspect the difference in signal strength and quality between them as the most probable factor.

I suspect it is down to the default TX power on the ESP32-Wroom. Im desperately struggling to find the right code to modify the firmware. It looks like the default TX for the Microbit is quite powerful

etmatrix, 2022-03-05:
The default tx power is 3dBm and the max should be 9 dBm.
I increased the power of TX with this patch

```

diff --git a/Firmware/ESP32/main/openhaystack_main.c b/Firmware/ESP32/main/openhaystack_main.c
index 7459e50..d5b0d3a 100644
--- a/Firmware/ESP32/main/openhaystack_main.c
+++ b/Firmware/ESP32/main/openhaystack_main.c
@@ -131,6 +131,8 @@ void app_main(void)
     esp_bluedroid_init();
     esp_bluedroid_enable();
 
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
+
     // Load the public key from the key partition
     static uint8_t public_key[28];
     if (load_key(public_key, sizeof(public_key)) != ESP_OK) {
```

With the app BLEScanner on android RX improve about 7/8 dBm