seemoo-lab/openhaystack #100: App displays "New version available" everytime

felixmueller, 2022-02-20:
**Describe the bug**
The message is displayed everytime on app launch, even if I already have the latest version (0.5.1)

**To Reproduce**
Steps to reproduce the behavior:
1. Lanuch App
2. See message

**Expected behavior**
Display message only if an update is available

**Screenshots**
[image]

**OpenHaystack version:**
0.5.1

**macOS version:**
12.2.1 (21D62) (Monterey)

mowtschan, 2022-02-21:
I have the same issue.

**OpenHaystack version:**
0.5.1 (1)

**macOS version:**
12.2.1 (21D62) (Monterey)

vvzvlad, 2022-05-04:
Confirming the problem
0.5.1,  MacOS 12.3.1 (21E258)

dchristl, 2022-05-08:
I had the same problemand there is an error in the code causing this issue. The UpdateCheckController checks the current version with the key CFBundleVersionShortString, but this one does not exist in Info.plist. There is one with name CFBundleShortVersionString (Version and Short swapped). I added the key from UpdateCheckController with:
`defaults write /Applications/OpenHaystack.app/Contents/Info CFBundleVersionShortString '0.5.1'` 
and the message was gone.

bluegizmo83, 2022-06-05:
Same issue