seemoo-lab/openhaystack #101: Restore the accessories from a backup

mowtschan, 2022-02-21:
I created a full backup from my old macbook with SuperDuper! app.
Is it possible to restore the accessories from that backup?

Thanks!

Sn0wfreezeDev, 2022-02-21:
The easiest way would be to export the accessories and import them in the app. 
The accessories are currently stored in the macOS keychain, but I am no expert in restoring keychain items from a backup, maybe keychain access offers such a feature? 
```
kSecAttrLabel: "FindMyAccessories",
kSecAttrService: "SEEMOO-FINDMY",
```

mowtschan, 2022-02-21:
I was able to import `FindMyAccessories` item with the native keychain tool! Works! Thanks!

closed 2022-02-21 by mowtschan

digitaltrousers, 2022-07-22:
> The easiest way would be to export the accessories and import them in the app.

I'm a bit confused by all the different keys. Could you please describe which ones to export in order to back up?
Thanks.

digitaltrousers, 2022-07-28:
Never mind, I found the import and export buttons in the app (box with arrow icons in top-left corner).