seemoo-lab/openhaystack #103: Unable to activate Mail Plugin

crazytil, 2022-02-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate the plugin and restart and error shows up like the screenshot shows

**OpenHaystack version:**
Version 0.5.1 (1)

**macOS version:**
12.3
Supported UUID Mail app: A4B49485-0377-4FAB-8D8E-E3B8018CFC21 

**Additional context**
[image]

PrinzEric, 2022-02-28:
We can only Still way for Update the UUID Springs ... THANKS

crazytil, 2022-02-28:
Hi @PrinzEric,
I have tried to do that with Xcode but get an error that the plugin is damaged

PrinzEric, 2022-02-28:
i think in the mailbundle info Change is easier

crazytil, 2022-02-28:
Any ideas or info on how to do this?

PrinzEric, 2022-02-28:
nope, i search for other plugins and infos how i can do