seemoo-lab/openhaystack #109: Support for MacOS Mail 16.0

klinquist, 2022-03-15:
**Describe the bug**
The OpenHaystack MacOS Mail plugin is not compatible with Mail 16.0, which ships with macOS 12.3

**OpenHaystack version:**
Version 0.5.1 (1)

**macOS version:**
12.3

dotsam, 2022-03-15:
I can confirm that adding this UUID to the `Supported12.3PluginCompatibilityUUIDs` key in the MailBundle's `Info.plist` file and then ad-hoc code signing it again has the bundle working correctly on 12.3.

felixmueller, 2022-03-17:
@dotsam Thank you for the hint!

I can confirm that it works by:

1. Disabling GateKeeper
`sudo spctl --master-disable`

2. Adding the UUID to the **Supported12.3PluginCompatibilityUUIDs** key in the MailBundle's **Info.plist** file and ad-hoc signing it with
`codesign --force --deep -s - OpenHaystackMail.mailbundle`

3. Re-activating the OpenHaystack Mail plugin checkbox in Mail.app

4. Re-enabling GateKeeper
`sudo spctl --master-enable`

The remaining problem is the hint to install and activate the mail plugin in the OpenHaystack app on every app launch:
[image]
Otherwise it seems to work fine...

0raculo, 2022-03-18:
Can't get this to work. Added the same UUID but I get the same error after restoring gatekeeper and restarting mail app.

Kherrisan, 2022-03-20:
The openhaystack mail plugin still could not be activated after my adding the new UUID to the info.plist of mailbundle. It told me it is not compatible.

**OpenHaystack version:**
Version 0.5.1 (1)
**Mail version:**
16.0 (3696.80.82.1.1)
**macOS version:**
12.3

basanys, 2022-03-21:
My /Library/Mail/Bundles folder is empty to try editing UUID.
What does above reference "MailBundle's Info.plist" folder mean?

rogueops, 2022-03-22:
It worked for me. @Kherrisan, try the attached, i added the string and signed it on my machine and it worked for me

https://mega.nz/file/IcUyELAR#lUfLvumMaxdXhwFN4JmV_3HRGPhlsfMoeYFjKIIotYU

jasonc62492, 2022-04-22:
How do us muggles get codesign?

rogueops, 2022-04-22:
look at the post above with the steps from felixmueller

jasonc62492, 2022-04-22:
> look at the post above with the steps from felixmueller

I dont have `codesign` available in my terminal

noahwilliamsson, 2022-04-23:
@jasonc62492 Perhaps you need to install the Xcode command line developer tools with the following command.

```
xcode-select --install
```

taitleaney, 2022-05-09:
I'm having trouble finding the info.plist file for the plugin. Where should I be looking?

noahwilliamsson, 2022-05-09:
@taitleaney Under `Library/Mail/Bundles/OpenHaystackMail.mailbundle/Contents/Info.plist` (relative to your user's home directory).

taitleaney, 2022-05-10:
Hmm, I've tried the steps suggested here, and I'm still getting the issue of the plugin being disabled. I've tried @noahwilliamsson 's commit, and the file provided by @rogueops with no luck. What should I try next?

Furtivo360, 2022-05-17:
Hi! I've tried to manually modify my `Info.plist` by addeding the UUID code needed and signing the file. I've also tried to use the files provided by our friend @rogueops but nothing. It's not working.
I can see the green dots next to the deployed tags but not the position on the map. 

[image]

And I am getting this message: 

[image]

What can I do?
Thanks!

wtfloris, 2022-05-18:
For anyone who is having issues (I got an error from Mail saying the plugin was incompatible):

Follow the steps by @felixmueller, but replace the 12.3 in **Supported12.3PluginCompatibilityUUIDs** by your OS version (latest being 12.4).

John-Kilo, 2022-05-31:
> My Info.plist seems to be gibberish. I've tried opening it with Atom and Visual Studio Code. Any suggestions on an editor I could use to make these changes?

I double tapped it, but it also works in xcode

closed 2022-06-01 by Sn0wfreezeDev

John-Kilo, 2022-06-02:
> completed via #114 <https://github.com/seemoo-lab/openhaystack/pull/114>.

Thank you so much, is there a way for me to donate or further support this
project?

pawisoon, 2022-06-03:
Sorry for reopening, but I have the same issue. I'm on MacOS 12.4 (21F79) and Mail.app version 16.0 (3696.100.31). I tried all solutions in this thread and "Manage plug-ins" is not visible. Did someone find a fix for it? Where can I find UUID that will work?

vvzvlad, 2022-06-07:
MacOS 12.4 recently came out, and apparently the plugin needs a new UUID because it is not visible in Mail.

noahwilliamsson, 2022-06-07:
@vvzvlad The most recent release (0.5.2) should be compatible with Mail.app on 12.4.  I just downloaded it, unzipped it, right-clicked it and selected "Open" and let it install itself.

Try copy-pasting the following into Terminal.app.

```
sudo defaults write "/Library/Preferences/com.apple.mail" EnableBundles 1
export B="$HOME/Library/Mail/Bundles/OpenHaystackMail.mailbundle"
sudo spctl --add --label 'Allow OpenHaystack Mail.app plugin to load' "$B"
```

If you get "Operation not permitted" while attempting to access files below `~/Library/Mail` in Terminal.app, then go to System preferences > Security & Privacy > Privacy, scroll down and select _Full disk access_.  Click _+_ and add `/Applications/Utilities/Terminal.app`.

After restarting Terminal.app, you should be able to access files below ~/Library/Mail.  Rerun the above commands and the "Operation not permitted" error should be gone.

If you restart Mail.app, you should find _Manage Plug-ins..._ in Mail.app preferences.  In there, check OpenHaystackMail.mailbundle, allow it to access files and then restart Mail.app again.

p1337, 2022-06-09:
Okay, this is not working any more on macOS Ventura 13.0 Beta (22A5266r) . 
Do you have to Update it for every version.?

John-Kilo, 2022-06-09:
Yes, it broke when I went from 12.3 to 12.4 monterey. Also, don't expect
support soon since the MacOs is just a beta version.

noahwilliamsson, 2022-06-09:
@p1337 You can try downloading and running the script at https://gist.github.com/noahwilliamsson/08500bb6417015b7d6023e0b42cac34e and see if that helps.  Feel free to share the output lines containing the _macOS version_ and _Mail.app plugin compatibility UUID_. It might be helpful for others running the beta.

p1337, 2022-06-12:
Hey folks,

```
macOS version (major, minor):       13.0
Mail.app plugin compatibility UUID: 890E3F5B-9490-4828-8F3F-B6561E513FCC
```

Mai bundle claims to support the following plugin compatibility UUIDs:
Print: Entry, "Supported13.0PluginCompatibilityUUIDs", Does Not Exist 
I will update the Info.plist - and try it again.

p1337, 2022-06-12:
Okay, I think, something went wrong .
after editing Info.plist - and signing it with:

```
codesign --force --deep -s - OpenHaystackMail.mailbundle
```

the Mail app, say, the File was created with no Date -> Move it to Trash ? Yes No Maybe.... blabla

Furtivo360, 2022-06-12:
any update?...I would be intrested to try your Info.plist if you've been able to create a new one...

Thanks! :)

olivluca, 2022-06-17:
Following the instructions in https://github.com/seemoo-lab/openhaystack/issues/109#issuecomment-1148735705 I could see the "Manage plugins" button and enable it, but then Mail.app crashes every time.
Also, every time I open OpenHaystack it tells me there's a new version (using 0.5.2), if I let it download the supposed new version and then open it it tells me that there is a new version (and so on).

MoutardeForte, 2022-10-19:
Same for me (mail v16 macOS 12.6)

jkangogo, 2022-11-15:
I there any latest update for macOS 13.0 or the latest?

HiAppHere, 2023-01-14:
[image]
see here #172

beckmx, 2023-01-18:
I have ventura 13.1 and also not working

tucka, 2023-02-20:
Same here. It's not working with Ventura 13.2.

What would be the right UUID to add?

Thanks!

kerta1n, 2023-02-21:
Same issue as well lol. New update screen every single time, plugin is "incompatible" on macOS 13.2, mail 16.0

Ret-Marut, 2023-03-14:
Adding this to the Info.plist worked at least on macOS 13.2.1:
[image]

PR #114 (2022-06-01): OpenHaystackMail: add support for Mail.app version 16.0 (macOS 12.3)
This updates the mail bundle's `Info.plist` with the expected Mail.app plugin compatibility UUID for Mail version 16.0 for macOS 12.3 and later.

Hopefully this should fix #109.  I'm not sure if this will require updated code signatures somewhere as well but to get things running locally, I used the following script: https://gist.github.com/noahwilliamsson/08500bb6417015b7d6023e0b42cac34e
commit 9829d6ce (2022-06-01): OpenHaystackMail: add support for Mail.app version 16.0 (macOS 12.3)

Fixed #109.