seemoo-lab/openhaystack #110: Possible to access timestamp in openhaystack app?

kafza, 2022-03-30:
HI,

I have been reading the published article, and checking through the code, and it seems like the timestamp for each datapoint is fetched from the iCloud server, and is used for the device location history option. 
However, I don't see any option for seeing the specific time stamp of each location point. Is it possible to access this on the openhaystack app? 

My second question is, how long are the location data stored on the iCloud server? Is it possible to go further back than 7 days?

bluegizmo83, 2022-06-12:
Adding to this, it would be nice if we could click on the datapoint pins in the Mac OpenHayStack app to see exactly what time that ping happened. As you said, that data is already there, as we can slide the timeframe slider to see past pings. We just need a way to see the data when clicking on a specific location pin...

jrusi, 2022-09-07:
That would be amazing! Maybe we can find it in some file locally? At least before it will be done in the app.