seemoo-lab/openhaystack #111: Tracking airtag

satishkn, 2022-03-31:
Reviewing other comments and question I have realized that openhaystack cannot be used to track existing airtags. If may ask a general question - any thoughts on how to track locations of existing airtags using any other way? Thanks in advance.

das800, 2022-06-12:
just to get this clearly. you want the OHS front end to show you genuine airtags as well as OHS devices? if you dont mind me asking, what is the usecase for this?

egrama, 2022-06-12:
I would like such a feature to be able to save the location history of a real airtag.
Something similar to the list of points on the map described in https://doi.org/10.2478/popets-2021-0045, but with a real airtag

luke-jr, 2022-06-12:
> just to get this clearly. you want the OHS front end to show you genuine airtags as well as OHS devices? if you dont mind me asking, what is the usecase for this?

Tracking genuine airtags. As things stand, I'm faced with a decision to either return the airtags I bought and go back to the drawing board (OHS-supported devices aren't very real-world-friendly), or buy an iOS device just to turn them on.

das800, 2022-06-14:
but why would you want to use a genuine airtag when this repo allows you to make airtags from cheap bluetooth microcontrollers?

luke-jr, 2022-06-14:
Fancy case that is more toddler-proof (and smaller?) than a bare microcontroller.

SchorschKloni, 2022-06-16:
Hello and good morning,
thanks @satishkn for this question. I thought the option to track real airtags is implementet in OHS.
So i wanted to work on the list to get it startet.
Now when i now, they are not, i saved a lot time.
Is here anybody able to pull lat and lon from the airtags with an IOs device?
I just have to put them in an URL to send it to the traccar server.
I would pay for that....

And @das800 : Airtags are very easy to use, and not very expensive

Sn0wfreezeDev, 2022-06-21:
We are already discussing these issues in  #97

closed 2022-06-21 by Sn0wfreezeDev